MarlinFirmware/Marlin #10342: Edge Not Working Properly with Auto Bed Levelling

zafff, 2018-04-07:
### Description
When "#define MIN_PROBE_EDGE 10" is set along with "#define AUTO_BED_LEVELING_BILINEAR", only x is working properly. The Y axis is left unchanged and is not taking the edge into consideration.
Suppose my bed is 230x230, and I am probing 9 points, the first one has X = 10 but the last point has coordinates X = 220 and Y = 230.

**Expected behavior:** 
I would expect that the last point to be X = 220 and Y = 220 as well.

**Actual behavior:** [What actually happens]
See description.

#### Additional Information
I am using a coreXY machine so it might be possible that this bug affects only coreXY and not other coordinate system.

Roxy-3D, 2018-04-08:
Which bed leveling system are you using?

zafff, 2018-04-08:
Bilinear.

// @section calibrate

/**
 * Choose one of the options below to enable G29 Bed Leveling. The parameters
 * and behavior of G29 will change depending on your selection.
 *
 *  If using a Probe for Z Homing, enable Z_SAFE_HOMING also!
 *
 * - AUTO_BED_LEVELING_3POINT
 *   Probe 3 arbitrary points on the bed (that aren't collinear)
 *   You specify the XY coordinates of all 3 points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - AUTO_BED_LEVELING_LINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a single tilted plane. Best for a flat bed.
 *
 * - AUTO_BED_LEVELING_BILINEAR
 *   Probe several points in a grid.
 *   You specify the rectangle and the density of sample points.
 *   The result is a mesh, best for large or uneven beds.
 *
 * - AUTO_BED_LEVELING_UBL (Unified Bed Leveling)
 *   A comprehensive bed leveling system combining the features and benefits
 *   of other systems. UBL also includes integrated Mesh Generation, Mesh
 *   Validation and Mesh Editing systems.
 *
 * - MESH_BED_LEVELING
 *   Probe a grid manually
 *   The result is a mesh, suitable for large or uneven beds. (See BILINEAR.)
 *   For machines without a probe, Mesh Bed Leveling provides a method to perform
 *   leveling in steps so you can manually adjust the Z height at each grid-point.
 *   With an LCD controller the process is guided step-by-step.
 */
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

/**
 * Normally G28 leaves leveling disabled on completion. Enable
 * this option to have G28 restore the prior leveling state.
 */
#define RESTORE_LEVELING_AFTER_G28

/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
//#define DEBUG_LEVELING_FEATURE

#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_BILINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  #define ENABLE_LEVELING_FADE_HEIGHT

  // For Cartesian machines, instead of dividing moves on mesh boundaries,
  // split up moves into short segments like a Delta. This follows the
  // contours of the bed more closely than edge-to-edge straight moves.
  #define SEGMENT_LEVELED_MOVES
  #define LEVELED_SEGMENT_LENGTH 5.0 // (mm) Length of all segments (except the last one)

  /**
   * Enable the G26 Mesh Validation Pattern tool.
   */
  //#define G26_MESH_VALIDATION
  #if ENABLED(G26_MESH_VALIDATION)
    #define MESH_TEST_NOZZLE_SIZE    0.4  // (mm) Diameter of primary nozzle.
    #define MESH_TEST_LAYER_HEIGHT   0.2  // (mm) Default layer height for the G26 Mesh Validation Tool.
    #define MESH_TEST_HOTEND_TEMP  205.0  // (°C) Default nozzle temperature for the G26 Mesh Validation Tool.
    #define MESH_TEST_BED_TEMP      60.0  // (°C) Default bed temperature for the G26 Mesh Validation Tool.
  #endif

#endif

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  #define GRID_MAX_POINTS_X 3
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  // Set the boundaries for probing (where the probe can reach).
  //#define LEFT_PROBE_BED_POSITION MIN_PROBE_EDGE
  //#define RIGHT_PROBE_BED_POSITION (X_BED_SIZE - MIN_PROBE_EDGE)
  //#define FRONT_PROBE_BED_POSITION MIN_PROBE_EDGE
  //#define BACK_PROBE_BED_POSITION (Y_BED_SIZE - MIN_PROBE_EDGE)

  // Probe along the Y axis, advancing X after each column
  //#define PROBE_Y_FIRST

  #if ENABLED(AUTO_BED_LEVELING_BILINEAR)

    // Beyond the probed grid, continue the implied tilt?
    // Default is to maintain the height of the nearest edge.
    //#define EXTRAPOLATE_BEYOND_GRID

    //
    // Experimental Subdivision of the grid by Catmull-Rom method.
    // Synthesizes intermediate points to produce a more detailed mesh.
    //
    //#define ABL_BILINEAR_SUBDIVISION
    #if ENABLED(ABL_BILINEAR_SUBDIVISION)
      // Number of subdivisions between probe points
      #define BILINEAR_SUBDIVISIONS 3
    #endif

  #endif

#elif ENABLED(AUTO_BED_LEVELING_UBL)

thinkyhead, 2018-04-08:
@zafff — Please ZIP and attach your `Configuration.h` and `Configuration_adv.h` files so we can see the rest of your settings.

zafff, 2018-04-08:
Surething. Please see attached.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1887708/Marlin.zip)

thinkyhead, 2018-04-09:
I can't account for your issue. Putting everything together it should all evaluate this way:
```cpp
#define X_BED_SIZE 230
#define Y_BED_SIZE 230

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 220

#define X_PROBE_OFFSET_FROM_EXTRUDER   0  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -36  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER  -1  // Z offset: -below +above  [the nozzle]

// Certain types of probes need to stay away from edges
#define MIN_PROBE_EDGE 10

#define _MIN_PROBE_X (max(0 + 10, 0 + 0))      	//  10
#define _MIN_PROBE_Y (max(0 + 10, 0 - 36))		  //  10
#define _MAX_PROBE_X (min(230 - 10, 230 + 0))	  // 220
#define _MAX_PROBE_Y (min(230 - 10, 230 - 36))  // 194

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)
  #ifndef LEFT_PROBE_BED_POSITION
    #define LEFT_PROBE_BED_POSITION 10
  #endif
  #ifndef RIGHT_PROBE_BED_POSITION
    #define RIGHT_PROBE_BED_POSITION 220
  #endif
  #ifndef FRONT_PROBE_BED_POSITION
    #define FRONT_PROBE_BED_POSITION 10
  #endif
  #ifndef BACK_PROBE_BED_POSITION
    #define BACK_PROBE_BED_POSITION 194
  #endif
#endif

inline bool position_is_reachable(const float &rx, const float &ry) {
  // Add 0.001 margin to deal with float imprecision
  return WITHIN(rx, 0 - 0.001, 230 + 0.001)
      && WITHIN(ry, 0 - 0.001, 230 + 0.001);
}

inline bool position_is_reachable_by_probe(const float &rx, const float &ry) {
  return position_is_reachable(rx - (0), ry - (-36))
      && WITHIN(rx, 0 - 0.001, 220 + 0.001)
      && WITHIN(ry, 0 - 0.001, 194 + 0.001);
}
```

So, the next thing to do is to get a log of your probing behavior.

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed.
- Copy the log output into a **.TXT file** and attach it to your next reply.

zafff, 2018-04-11:
I attached the logs of G28 and G29 in 2 separate files. I updated to the latest bugfix (10 April 18).

[G28.txt](https://github.com/MarlinFirmware/Marlin/files/1896963/G28.txt)
[G29.txt](https://github.com/MarlinFirmware/Marlin/files/1896964/G29.txt)

thinkyhead, 2018-04-11:
The `G29` log shows only 8 points. If you're doing a 3x3 grid (as in your configuration) then there should be 9. But actually it looks like you're doing a 4x4 grid, at least in the X dimension. The points in your log are shown as:
```
220.00, 132.00
150.00, 132.00
80.00, 132.00
10.00, 132.00
10.00, 193.00
80.00, 193.00
150.00, 193.00
220.00, 193.00
```
Everything appears to be within the expected bounds.

zafff, 2018-04-12:
I'm sorry, I changed the grid from 3x3 to 4x4 and the probe points are the following:
/**
10.00, 132.00
10.00, 168.00
10.00, 193.00
10.00, 229.00

80.00, 132.00
80.00, 168.00
80.00, 193.00
80.00, 229.00

150.00, 132.00
150.00, 168.00
150.00, 193.00
150.00, 229.00

220.00, 132.00
220.00, 168.00
220.00, 193.00
220.00, 229.00
**/

I think I might have confused the probe and the nozzle in this case as the nozzle reaches the point 229 in the Y direction but the probe is behind it in the Y direction. 
Sorry for the trouble. And thanks for your input. I am closing the issue.

closed 2018-04-12 by zafff

thinkyhead, 2018-04-12:
Thanks for the update!

inventabuild, 2018-09-22:
> So, the next thing to do is to get a log of your probing behavior.
> 
> * Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.

bugfix-2.0.x contains only some of the Marlin files.  Do I copy the downloaded bugfix-2.0.x files over the final release files of 1.1.9 to make a version of bugfix-2.0.x for testing?

thinkyhead, 2018-09-22:
@inventabuild — The files are there, but in subfolders. Instead of using Arduino you’ll need to use a proper project editor if you need to edit any of the code files in subfolders. Sublime Text, Notepad++, and Visual Studio Code are all popular choices.

inventabuild, 2018-09-22:
> @inventabuild — The files are there, but in subfolders. Instead of using Arduino you’ll need to use a proper project editor if you need to edit any of the code files in subfolders. Sublime Text, Notepad++, and Visual Studio Code are all popular choices.

Thanks and can I use Arduino to upload the firmware to my printer even if only a few of the files are showing there when I open Marlin.ino or do I need to use a different app for the upload?

netzmark, 2018-12-31:
Hi,
I hope I can continue this thread insted making new one as the title is good for my "issue".
I have JGAurora with 1.1.9 prepared by Samuel Pinches. Unfortunately I'm not sure which bugfix he applied but for sure not 2.0.

Anyway, my problem is following.
I have BLtouch mounted 6mm to the right and 36mm behind the nozzle (between nozzle and rear edge of the bed).
So I have declared: 
#define X_PROBE_OFFSET_FROM_EXTRUDER 6  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 36  // Y offset: -front +behind [the nozzle]

Looking on above conversation I'm not sure whether I should declare it + or -36. Zafff is telling he has "inductive sensor placed behind the nozzle" (as mine) but declaration is -36. Documentation says me rather it should be +. I'm not sure whether type of driving is important here, my printer has yz driven bed.

Problem is that when I perform UBL the left, right and rear margins are applied in accordance to the #define MIN_PROBE_EDGE. But front probes are begun from Y36, it means for the first raw the head is moved 36 forward from 0.

What could I do wrong?
Attaching my configuration.h
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2718357/Configuration.zip)

netzmark, 2018-12-31:
This is the first probe output:
SENDING:G29 P1
Mesh invalidated. Probing mesh.
  current_position=(-5.00, 0.00, 10.85) : set_probe_deployed
deploy: 1
do_probe_raise(10.00)
>>> do_blocking_move_to(-5.00, 0.00, 10.85)
<<< do_blocking_move_to
>>> probe_pt(135.22, 68.11, raise, 0, probe_relative)
  current_position=(-5.00, 0.00, 10.85) :
>>> do_blocking_move_to(129.22, 32.11, 10.85)

So the first test is exactly made at the Y32.11 and I cannot to force it to be performed at Y0...