MarlinFirmware/Marlin #10344: STM32F1 (MapleMini) - G1 Zx and G1 Xx both running the same motor

etet100, 2018-04-08:
### Description

Hi. Both G1 Zx and G1 Xx commands and running the same motor (X axis). Y axis
is working fine. My PINs configuration is:

```
#define X_STEP_PIN         15
#define X_DIR_PIN          16
#define X_ENABLE_PIN       -1
#define X_MIN_PIN          -1
#define X_MAX_PIN          -1

#define Y_STEP_PIN         18
#define Y_DIR_PIN          19
#define Y_ENABLE_PIN       -1
#define Y_MIN_PIN          -1
#define Y_MAX_PIN          -1

#define Z_STEP_PIN         14
#define Z_DIR_PIN          13
#define Z_ENABLE_PIN       -1
#define Z_MIN_PIN          -1
#define Z_MAX_PIN          -1

#define Y2_STEP_PIN        -1
#define Y2_DIR_PIN         -1
#define Y2_ENABLE_PIN      -1

#define X2_STEP_PIN        -1
#define X2_DIR_PIN         -1
#define X2_ENABLE_PIN      -1

#define Z2_STEP_PIN        -1
#define Z2_DIR_PIN         -1
#define Z2_ENABLE_PIN      -1

#define E0_STEP_PIN        10
#define E0_DIR_PIN         11
#define E0_ENABLE_PIN      12

```

I'm almost sure everything was fine a couple weaks ago. 

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1887184/configs.zip)

### Steps to Reproduce

1. Open Repetier Host
2. Connect
3. Move X axis
4. Move Z axis

**Expected behavior:** 

Moving X axis motor
Moving Z axis motor

**Actual behavior:** 

Moving X axis motor (slower)
Moxing X axis motor (faster)

#### Additional Information

Configuration files attached.

kAdonis, 2018-04-08:
Are you just testing, or do you really have a core xz printer?
how did you wire the ENABLE pins?

etet100, 2018-04-08:
Ok. I thought i have to uncomment at least one of these:
```
//#define COREXY
//#define COREXZ
//#define COREYZ
//#define COREYX
//#define COREZX
//#define COREZY
```
Now i see that cetus 3d is not a coreZX printer. I have commented it back and now
it's working as expected.

Thank you.

closed 2018-04-08 by etet100