MarlinFirmware/Marlin #10345: STM32F103CB(MALYAN), E0_DIR_PIN problem！！！ can not output high...

zlychen, 2018-04-08:
No matter how I set it, it can't output high level and switch the running direction of the extruder.

 Here's my configuration.

I tested the xyz directional control pins and all XYZ  switched normally. .

#define X_STEP_PIN         PB14
#define X_DIR_PIN          PB15
#define X_ENABLE_PIN       PA8

#define Y_STEP_PIN         PB12
#define Y_DIR_PIN          PB13
#define Y_ENABLE_PIN       PB9

#define Z_STEP_PIN         PB10
#define Z_DIR_PIN          PC13
#define Z_ENABLE_PIN       PB11

i change E0_DIR_PIN configuration in PC13 or PA2 and PB0, but did not change the result..

#define  E0_STEP_PIN        PB0
#define  E0_DIR_PIN         PB2//PC13
#define  E0_ENABLE_PIN      PB1

please help me to fixed this problem... thanks....

zlychen, 2018-04-09:
I found my problem, in #define in advance option。。i close this fuction will work E0_DIR_PIN ok...

but E0_DIR_PIN still have problem...  i doing more teste on E0_DIR_PIN... 

if i turn output high level on  XYZ DIR_PIN,  E0_DIR_PIN turn automatic of low level......

please help me to fixed this problem... thanks....

Eno423, 2018-04-13:
@xC0000005 Is this what you were talking about?

xC0000005, 2018-04-13:
Which model Malyan is this running on? On M200 V1 and V2 (all revs), it's PC13. PB2 should be the Z dir.

zlychen, 2018-04-14:
Thank xC0000005 for your reply。 i have only test the last version of marlin 2.0.x.. i know orginal define pin E0-DIR is PC13....   problem is if i turn output high level on XYZ DIR_PIN any one of them, E0_DIR_PIN turn automatic of low level...... Although I set E0_DIR_PIN high..

zlychen, 2018-04-14:
I do not have Malyan M200 V1 or V2。 I want to build a 3d printer based on bleupill(using malyan board define) . i have build my self a clone i3 3d printer and work good. and  i build test board( base on bluepill) and i test all pins. but have found the problem on E0-DIR pin.... i down know how to fix this problem.....  and i posted this issues.... 
please help me to fix this problem.... thanks a lot....

xC0000005, 2018-04-14:
Ok, I'm sorry - I was only commenting because I've been working on getting Marlin working on the Malyan boards and this had that in the title. I did have a bluepill wired to the RAMPS 1.4 setup for a while, and it worked ok. If you load a test program onto your board to test the different pins, does it step the motors correctly?

zlychen, 2018-04-14:
step motors on XYZ worked ok.. only E0-DIR-PIN is not worked ok. .  I use simplify3d to connect to the sd module with bluepill to simulate printing about 10 hours is doing good....  Anyway, thank you for your help..

closed 2018-07-04 by zlychen