MarlinFirmware/Marlin #10346: [1.1.8] Bed Skew compensation

anton-piliugin, 2018-04-08:
I'm using Marlin 1.1.8 latest release version.
Bed skew compensation configuration (default values):

```
#define SKEW_CORRECTION

#if ENABLED(SKEW_CORRECTION)
  // Input all length measurements here:
  #define XY_DIAG_AC 282.8427124746 //140.5
  #define XY_DIAG_BD 282.8427124746 //141
  #define XY_SIDE_AD 200 //99.5

  // Or, set the default skew factors directly here
  // to override the above measurements:
  //#define XY_SKEW_FACTOR 0.0

  //#define SKEW_CORRECTION_FOR_Z
  #if ENABLED(SKEW_CORRECTION_FOR_Z)
    #define XZ_DIAG_AC 282.8427124746
    #define XZ_DIAG_BD 282.8427124746
    #define YZ_DIAG_AC 282.8427124746
    #define YZ_DIAG_BD 282.8427124746
    #define YZ_SIDE_AD 200
    #define XZ_SKEW_FACTOR 0.0
    #define YZ_SKEW_FACTOR 0.0
  #endif

  // Enable this option for M852 to set skew at runtime
  #define SKEW_CORRECTION_GCODE
#endif
```
I got compilation error:
```
In file included from sketch\MarlinConfig.h:38:0,

                 from sketch\G26_Mesh_Validation_Tool.cpp:27:

macros.h:218: error: 'sqrt(2.91072e+5)' is not a constant expression

 #define SQRT(x)     sqrt(x)

                           ^

C:\Users\Anton\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino/Arduino.h:99:17: note: in definition of macro 'sq'

 #define sq(x) ((x)*(x))

                 ^

sketch\Conditionals_post.h:856:33: note: in expansion of macro '_SKEW_SIDE'

     #define _SKEW_FACTOR(a,b,c) _SKEW_SIDE(a,_GET_SIDE(a,b,c),c)

                                 ^

sketch\Conditionals_post.h:854:31: note: in expansion of macro 'SQRT'

     #define _GET_SIDE(a,b,c) (SQRT(2*sq(a)+2*sq(b)-4*sq(c))*0.5)

                               ^

sketch\Conditionals_post.h:856:46: note: in expansion of macro '_GET_SIDE'

     #define _SKEW_FACTOR(a,b,c) _SKEW_SIDE(a,_GET_SIDE(a,b,c),c)

                                              ^

sketch\Conditionals_post.h:861:11: note: in expansion of macro '_SKEW_FACTOR'

           _SKEW_FACTOR(XY_DIAG_AC, XY_DIAG_BD, XY_SIDE_AD)

           ^

exit status 1
'sqrt(2.91072e+5)' is not a constant expression
```

thinkyhead, 2018-04-09:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

anton-piliugin, 2018-04-09:
Looks like it has been fixed in bugfix-1.1.x

closed 2018-04-09 by anton-piliugin