MarlinFirmware/Marlin #10352: G29 problem on a flsun delta

peepsquak, 2018-04-09:
Hello, 
Recently I bought a delta 3d printer from flsun on aliexpress. 
I received the printer as a DIY, so I built it, and I had no special problems. 
Now when I'm on REPETIER and when I lauch the autolevelling by the g29 command, the nozzle go a little and start beging the process in the air, I tried many things to solve that, but every times i believed now it's ok, it's always levelling in the air. 
Please someone can help me ? 
Thanks.

joris57, 2018-04-09:
For support of FlSun printers, you may go here : https://www.facebook.com/groups/FLSUN3DP/

Roxy-3D, 2018-04-09:
You do realize there are 5 different (and distinct) bed leveling systems, and each of them has a G29 command?

psavva, 2018-04-10:
Did you compile Marlin yourself?  Out are you using their stock firmware?

thinkyhead, 2018-04-10:
Either your probe is malfunctioning (maybe connected to the wrong pins) or the firmware is mis-configured for your specific probe. First check all your wiring to make sure it is correct.

After that, are you prepared to configure and flash a new Marlin onto your machine? If so, download the latest `bugfix-1.1.x` and you'll need to redo your configuration starting from one of the included example configurations for the FLSUN delta, located in `Marlin/example_configurations/delta/FLSUN`.

peepsquak, 2018-04-10:
Thanks all for your answer. 😉
@joris57 I don't have a facebook account. 
@psavva I'm using their marlin firmware. 
I think I don't have problems with the probe. 
I use the M119 after a G28 while I was pushing the nozzle. Here is the results : 

16:31:12.593 : Reporting endstop status
16:31:12.593 : x_max: TRIGGERED
16:31:12.593 : y_max: TRIGGERED
16:31:12.597 : z_min: TRIGGERED
16:31:12.597 : z_max: TRIGGERED

16:31:22.410 : Reporting endstop status
16:31:22.410 : x_max: TRIGGERED
16:31:22.410 : y_max: TRIGGERED
16:31:22.414 : z_min: open
16:31:22.414 : z_max: TRIGGERED

thinkyhead, 2018-04-10:
We cannot give support for third-party forks. The better place to ask these kinds of questions is [RepRap.org Forums](http://forums.reprap.org/).

joris57, 2018-04-10:
@peepsquak Until yesterday, you didn't have a github account either ...

I was just pointing out there is a quite thriving and friendly FlSun community there, that may be able to help you and is much more suited to support requests than this platform (as explained by thinkyhead).

ChrisP28, 2018-04-11:
Hi,I have had this problem. I found that when using the Allen key type Z probe it would bind in the aluminium print head assembly. It was never consitant when the probe deployed. I also found this probe type jammed during the level probing routine causing some spectacular bed / probe clashes. My solution so far is to replace the Allen key probe with a proximity switch permenently mounted on the print head assembly. I found that a magnetic type switch was ineffective so switched to a capacitance type running at 24Vdc (gives a greater detection range than 12V) through a relay to trigger the Z min input. It worked fine until I messed with the software but that’s another story.

boelle, 2019-02-20:
@peepsquak 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

peepsquak, 2019-02-20:
🖕

boelle, 2019-02-20:
@peepsquak then please hit the green close bottom below

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead