MarlinFirmware/Marlin #10354: Z Probe failed to retract on proximity probe.

ChrisP28, 2018-04-09:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* Read the "Contributing" page (under "Helpful resources") for additional guidelines.

psavva, 2018-04-10:
Any details on the issue would be helpful, as well as your configuration files

closed 2018-04-10 by thinkyhead

ChrisP28, 2018-04-10:
I work away so I’ll have to get the configuration files when I return. Thanks

thinkyhead, 2018-04-10:
Great. Please start a new issue and fill out all the requested information.