MarlinFirmware/Marlin #10355: corexy: INVERT_#_DIR changes motor direction instead direction of movement

hafpel, 2018-04-09:
### Description

Marlin: origin/bugfix-1.1.x
Printer: Tronxy x5s

Right now I've found no way to change the direction of a single axis with a COREXY: the INVERT_#_DIR flags invert the motor movement, but in a COREXY this means that both axis get inverted, not a single one. his is a bit of a nuisance as my printer right now sports a "left" coordinate system which is sort of a nuisance if you print nonsymmetric parts.
The endstop reference is sort of related: as the coordinates turn 90° with each inversion while the allocation of the endstops to the axis stays the same the head will crash if a single motor is inverted as the x movement triggers the y endstop.

### Steps to Reproduce

If you have any corexy, try to invert a single movement axis by inverting a motor direction in configuration.h line 765 ff

**Expected behavior:** [What you expect to happen]

INVERT_#_DIR inverts the axis on a COREXY

**Actual behavior:** [What actually happens]

INVERT_#_DIR inverts the motor direction

#### Additional Information

---

KevMags, 2018-04-09:
Change firmware and wiring to change your endstop to a MAX endstop instead of a MIN endstop .  That will invert the coordinate system of that axis.  For example,  #define USE_XMAX_PLUG instead of #define USE_XMIN_PLUG (and change endstop connector to XMAX pins on controller).  Change homing direction as needed (X_HOME_DIR 1 is used to home to max on typical installation).

hafpel, 2018-04-09:
While this serves problem that the endstop switch is on the "wrong side" it doesn't change the direction of the coordinate system itself which I need to flip over.

hafpel, 2018-04-09:
If you change the endstop from #define USE_XMIN_PLUG  to #define USE_XMAX_PLUG and change the direction by X_HOME_DIR 1 as well all it does is to move in the other direction and use the stop osition as maximum - which really does nothing to the axis (tried it some minutes ago). What you probably did is to change both axis and endstops so you basically rotated your coordinate system 180° but didn't changed the "hardness" (sorry, I'm from germany and here it's called a "right-hand" vs "left-hand" system: thumb is x, forefinger y and middlefinger z-axis).

What did help was to switch the motor cables X<->Y: now it's a right hand system and it works. SO beside the nuisance with the flipped endstops I'd call the case closed

closed 2018-04-09 by hafpel

Spiritdude, 2019-01-01:
If X axis is reversed, following is the way to do (after many attempts):
- switch cables A & B (defining COREYX instead COREXY won't help) but keep `#define COREXY`
- reverse values of INVERT_X_DIR and INVERT_Y_DIR (if they were false => true, or if they were true => false)

The preview suggestions of HOME_DIR_X or endstops location **does not affect direction** of the axis (I just tested it on my system). So, @hafpel suggestion of just swapping cables is the way to go.

pkucmus, 2019-11-20:
@Spiritdude Thanks it worked like a charm - I was loosing my mind :)