MarlinFirmware/Marlin #10356: hot end thermistor 300 celsius + hotend does not want to heat up

REVIEWGAMINGMAN, 2018-04-09:
[image]

ManuelMcLure, 2018-04-09:
This is most likely a hardware issue with the hotend thermistor input on your motherboard or an incorrect thermistor selection in your configuration.

REVIEWGAMINGMAN, 2018-04-10:
would reflashing the bootloader help

ManuelMcLure, 2018-04-10:
I would very much doubt it.

thinkyhead, 2018-04-10:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

REVIEWGAMINGMAN, 2018-04-11:
[
[image]
](url)

REVIEWGAMINGMAN, 2018-04-11:
[image]

Grogyan, 2018-04-11:
Thermistor is damaged and needs replacing

boelle, 2019-02-20:
@REVIEWGAMINGMAN did you try replacing thermistor?

@thinkyhead think we can close this one

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead