MarlinFirmware/Marlin #10359: [bugfix-1.1.x] SanityCheck X_MIN_ENDSTOP_INVERTING

Ludy87, 2018-04-09:
### Description

I try to activate SENSORLESS_HOMING and get the error.

       SanityCheck.h:1446: error: #error "SENSORLESS_HOMING requires X_MIN_ENDSTOP_INVERTING when homing to X_MIN."

       #error "SENSORLESS_HOMING requires X_MIN_ENDSTOP_INVERTING when homing to X_MIN."

### Steps to Reproduce

1. activeate SENSORLESS_HOMING in Configuration_adv.h
2. compiling

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1891676/Configuration.zip)

sorry for my bad English

psavva, 2018-04-10:
In your configuration.h file, change
#define X_MIN_ENDSTOP_INVERTING false
to
#define X_MIN_ENDSTOP_INVERTING true

thinkyhead, 2018-04-10:
Yep! We often use `#error` to tell you what you need to change.

Ludy87, 2018-04-10:
When set to true, the ENDSTOP is triggered and is not open!

marcio-ao, 2018-04-10:
Do you have the pullup resistors enabled?

Ludy87, 2018-04-10:
No, why?

marcio-ao, 2018-04-10:
@Ludy87 : I'm just trying to narrow down the problem. If the endstops had not been enabled, it would explain why sensorless homing was not working for you.

Ludy87, 2018-04-10:
Real? Before have i update install, no error!

marcio-ao, 2018-04-10:
@Ludy87 : I believe you when you say something is wrong, I just don't know what.

thinkyhead, 2018-04-12:
@Ludy87 — Try `M502` followed by `M500` and re-start the machine to see if it helps.

thinkyhead, 2018-04-12:
@marcio-ao — The line that was removed…. Should it perhaps just have been _changed_ instead?

marcio-ao, 2018-04-12:
@thinkyhead : I could have changed it to the following:

`st.diag1_active_high(0); // For sensorless homing`

However, this is supposed to be the power up default. I suppose people could be having issues if they upgraded their FW and did not cycle the power on their printers.

Ludy87, 2018-04-13:
sorry, it was a short on the board. Apologize a lot!

closed 2018-04-13 by Ludy87