MarlinFirmware/Marlin #10367: [FR] Feature to keep Z-motors energized on Z-belt driven printers

marcio-ao, 2018-04-10:
Hello,

We are working on a Z-belt driven printer and for this, we need to keep the Z motors energized to keep the hot-end from falling on the print. I've set DISABLE_INACTIVE_Z to false, but there are still instances where this feature does not do enough. Specifically:

- We want the stepper at startup as soon as possible (we have an electrical motor brake, but it releases as soon as the printer is powered on).
- Marlin seems to de-energize the steppers at the end of an SD card print.
- It would be nice if M18 would not disable Z unless it was specifically named.

Perhaps it makes sense to have a specific setting that goes beyond DISABLE_INACTIVE_Z and is meant specifically for printers in which disabling the Z stepper could cause the head to fall.

marcio-ao, 2018-04-10:
Ohh, okay, it looks like what was causing my steppers to deactivate was the following in `configuration_adv.h`:

```
  #define SD_FINISHED_STEPPERRELEASE true          // Disable steppers when SD Print is finished
  #define SD_FINISHED_RELEASECOMMAND "M84 X Y Z E" // You might want to keep the z enabled so your bed stays in 
```

No clue why these settings exist, but I guess this was the problem. So I think the only thing needed right now is to enable the Z at startup.

thinkyhead, 2018-04-13:
> No clue why these settings exist

They've been in Marlin from the beginning, so the reason is lost in the annals of time.

thinkyhead, 2018-04-13:
> I think the only thing needed right now is to enable the Z at startup.

That should be fairly easy to apply to your specific machine's code. Note that not every conceivable customization needs to be made into a new configuration option in the main project. It may be enough to add a more general feature, such as `STARTUP_GCODE`, that would run after init and EEPROM loading.

marcio-ao, 2018-04-13:
@thinkyhead : Would it make sense to automatically energize the motor at startup if DISABLE_INACTIVE_[XYZ] was defined? I can't really think of a reason why someone would want to disable the inactivation of idle motors, but be alright with them being inactive at startup.

thinkyhead, 2018-04-14:
> if `DISABLE_INACTIVE_[XYZ]` was defined? 

You mean, set `false`?

I don't know. I think most of the time it's better if we don't power up steppers right at startup, but others might have a different opinion. I can imagine that generating a lot of "bug reports."

marcio-ao, 2018-04-16:
> You mean, set false?

Yes. That's what I meant.

> I don't know. I think most of the time it's better if we don't power up steppers right at startup, but others might have a different opinion. I can imagine that generating a lot of "bug reports."

DISABLE_INACTIVE_[XYZ] is true by default, so the behavior would only change for people who used DISABLE_INACTIVE_[XYZ] false. I guess the question to ask is what's the use case people are going for by setting that to false and whether activating the steppers at startup would affect their use case. In our Z-belt driver printers, the use case is to keep the motors energized while the printer is on so the X-axis doesn't sag -- a relay brakes (shorts) the motors when the power is shut off -- so having the motors on all the time makes sense.

marcio-ao, 2018-04-16:
How about just giving folks the choice? The activation at start feature seems closely related to the DISABLE_INACTIVE_[XYZ] choices:

```
#define DISABLE_INACTIVE_X true
#define DISABLE_INACTIVE_Y true
#define DISABLE_INACTIVE_Z true
#define DISABLE_INACTIVE_E true
#define ACTIVATE_AT_BOOT_X false
#define ACTIVATE_AT_BOOT_Y false
#define ACTIVATE_AT_BOOT_Z false
#define ACTIVATE_AT_BOOT_E false
```

While I don't have a huge problem in doing that modification in our own FW, any of our custom changes must be moved over to any subsequent release, so there's a benefit in not having to do that. I also don't suspect this is a very intrusive change; all it will probably add is a couple #if ENABLED() to the Marlin startup code.

thinkyhead, 2018-04-20:
Sure, add a new set of options.

marcio-ao, 2019-02-20:
Decided to close this. Can be re-opened if someone else wants this feature. We have our own startup init function now that handles this and a couple other things.

closed 2019-02-20 by marcio-ao