MarlinFirmware/Marlin #10371: 1.1.8 Bugfix: Save Z-Offset from menu

insanity67, 2018-04-10:
Is there any way to save an z-offset (from babystepping) to the eeprom?

silentninja1, 2018-04-10:
Store settings saves the z offset along with any other changes under the control menu.

insanity67, 2018-04-10:
OK, I just read that I have to enable BABYSTEP_ZPROBE_OFFSET, but not having a real probe, I get following error on compile:
#error "BABYSTEP_ZPROBE_OFFSET requires a probe."

Just not possible then? Or are the settings saved anyway?

silentninja1, 2018-04-10:
Z offset and z probe offset are 2 different things that get converged when you have a probe and enable babystepping. If you dont have a probe, then leave it off and the z offset is saved as expected.

insanity67, 2018-04-10:
OK thanks for your support! Was quite confused about that...

closed 2018-04-10 by insanity67