MarlinFirmware/Marlin #10372: Seemingly random motion after "Stop print"

marcio-ao, 2018-04-10:
The "Stop Print" command from the LCD menu sometimes generates a seemingly random prior to stopping that can sometimes cause the printer to run into the limits.

1. Start a print from the LCD
2. Select "Stop Print"
3. Watch the hotend carefully.
4. Repeat this several times... it does not always happen.

**Expected behavior:** The printer should stop immediately.

**Actual behavior:** Every few times, the printer hotend will move in a seemingly random direction prior to stopping.

On the 1.1.8 stable branch, I see this behavior along with a momentary flash of the "Print Pause" screen. On the "bugfix 2.0.x" branch, it seems as if the LCD flash has been fixed, but the random motion after stop print remains.

thinkyhead, 2018-04-12:
I haven't seen this issue yet, but will do some testing soon. Preliminary questions:
- Does this only happen if the carriage is already moving?
- Does the carriage move in XY and/or Z, or just the extruder motor?

Any theories as to where this extra move is coming from?

marcio-ao, 2018-04-12:
I had a theory, but I've ruled it out. Some of the people at my company had been reporting this issue more than others. At first, I thought maybe it had to do with what model was being printed. I thought maybe that the error was being caused by some line of GCODE from the original model being left in the planner after a stop. This would mean that the error would be more noticeable on prints with small movements (such as benchy), while more noticeable on bigger parts with bigger movements. But I printed a big cube and in every time I tried it, the print seemed to stop on command. So that theory seems to be wrong and at the moment I do not know under what circumstances the extra moves crop up.

thinkyhead, 2018-04-13:
Interesting… Well hopefully a 100% reproducible case can be devised. I was looking to make sure the ISR is disabled as part of the process, and apparently it is.

marcio-ao, 2018-04-13:
@januar3829 : Yes, that looks like exactly it.

thinkyhead, 2018-04-14:
It has the appearance of the previous block getting repeated. Keep an eye on this effect and see if that is consistent. I've patched the bugfix branches with some redundancies that probably won't have any effect, but be sure to use the latest code anyway.

januar3829, 2018-04-14:
Sorry, I forgot to mention that this issue started when I updated from 1.1 to 1.1.8

thinkyhead, 2018-04-14:
@januar3829 — 1.1.0 ?

januar3829, 2018-04-14:
Yes, 1.1.0, it was the original Firmware on my Tevo Tornado

thinkyhead, 2018-04-16:
I've added a patch that does Stop Print a little differently. See if the current `bugfix-1.1.x` behaves any better.

thinkyhead, 2018-04-20:
> I don't want the Printer to cool down after I stopped a Print, where can I adapt that?

`ultralcd.cpp`:

```diff
  void lcd_sdcard_stop() {
    card.stopSDPrint();
    clear_command_queue();
    quickstop_stepper();
    print_job_timer.stop();
-   thermalManager.disable_all_heaters();
    #if FAN_COUNT > 0
      for (uint8_t i = 0; i < FAN_COUNT; i++) fanSpeeds[i] = 0;
    #endif
    wait_for_heatup = false;
    lcd_setstatusPGM(PSTR(MSG_PRINT_ABORTED), -1);
    lcd_return_to_status();
  }
```

marcio-ao, 2019-02-01:
Closing this ticket since I haven't seen this problem in a while.

closed 2019-02-01 by marcio-ao