MarlinFirmware/Marlin #10378: Fichier source Crealty3D CR-10S officiel

boubouexpress, 2018-04-11:
https://github.com/ShenzhenChuangxiang3D/Marlin/releases/tag/CR-10S--1.1.0-RC3

idanmos, 2018-04-12:
Will anyone think to import the 'Resume on Power-Loss (SDCard-based)' feature?
I think this can be great.

valerie16901, 2018-04-12:
Yes, please!
But it will be great to have an option to enable it or disable it from the menu, since it makes continuous heavy use of the SD Card during SD printing maybe not everyone would want it enabled.

thinkyhead, 2018-04-13:
> Will anyone think to import the 'Resume on Power-Loss (SDCard-based)' feature?

[image]

boelle, 2019-02-20:
@thinkyhead funny title/subject, what does it mean?

InsanityAutomation, 2019-10-15:
@boelle @shitcreek The needful from here has been incorporated. This can be closed.

closed 2019-10-15 by boelle