MarlinFirmware/Marlin #10379: Cannot compile TMC2208/SoftwareSerial DUE/RADDS

devians88, 2018-04-11:
Hi My problem!

Arduino: 1.9.0-beta (Windows 10), Board: "Arduino Due (Programming Port)"

C:\Users\Lapos\Documents\Arduino\libraries\SoftwareSerial-master\SoftwareSerial.cpp:260:2: error: #error This version of SoftwareSerial supports only 20, 16 and 8MHz processors

 #error This version of SoftwareSerial supports only 20, 16 and 8MHz processors

  ^

exit status 1
Error compiling for board Arduino Due (Programming Port).
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1900502/config.zip)

teemuatlut, 2018-04-11:
Arduino Due doesn't have an "official" SoftwareSerial library that can be utilized.
The only one that comes up when searching is soft_uart by antodom but it hasn't been implemented into the TMC2208Stepper library.

ZenMops, 2018-04-16:
Is there any workaround to get TMC2208 compiled for the Arduino Due?

teemuatlut, 2018-04-16:
Modify the TMC2208 library to use soft_uart.
I might look into this myself at some point but it'll take a while before I get to it.

boelle, 2019-02-20:
@devians88 still having issues if you use the latest bugfix 2.0?

closed 2019-07-11 by shitcreek