MarlinFirmware/Marlin #10381: hotend and extruder 2 heats together when i heat up bed or extruder(any one)

amrithmmh, 2018-04-12:
i am using ramps 1.4 and latest ramps 1.1.8 version.
when i heat my extruder 2 the bed automatically heats up and then the extruder 2 goes beyond the limit (180 for pla).

thinkyhead, 2018-04-12:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

ManuelMcLure, 2018-04-12:
Make sure that the metal tabs on the RAMPS MOSFETs are not touching. That can cause strange behavior where turning on one RAMPS output causes another RAMPS output to turn on as well. It's happened to me and I know of at least one other person with the same problem.

amrithmmh, 2018-04-13:
turns out my wires were bad and had to change them ..so its not marlin firmware issues so closing this issue ...thanks for the suggestion @ManuelMcLure

closed 2018-04-13 by amrithmmh