MarlinFirmware/Marlin #10382: TMC 2130 not working properly

craitor, 2018-04-12:
I am using TMC 2130 stepper drivers. However, my stepper motors are constantly disabled.

I am using the RAMPS 1.4 together with the repetier software. I have the most recent version of Marlin installed. The TMC2130 are from Watterrot.

I ran the M122 command in Repetier and the stepper motors are constantly disabled. Here is the output I obtained from the M122 command:

00:33:24.966 : X	Y	Z
00:33:24.966 : Enabled	false	false	false
00:33:24.966 : Set current	800	800	800
00:33:24.970 : RMS current	1436	1436	1436
00:33:24.970 : MAX current	2025	2025	2025
00:33:24.970 : Run current	25/31	25/31	25/31
00:33:24.974 : Hold current	12/31	12/31	12/31
00:33:24.974 : CS actual	0/31	0/31	0/31
00:33:24.974 : PWM scale	0	0	0
00:33:24.974 : vsense	0=.325	0=.325	0=.325
00:33:24.978 : stealthChop	false	false	false
00:33:24.978 : msteps	256	256	256
00:33:24.978 : tstep	0	0	0
00:33:24.978 : pwm
00:33:24.978 : threshold	0	0	0
00:33:24.982 : [mm/s]	-	-	-
00:33:24.982 : OT prewarn	false	false	false
00:33:24.982 : OT prewarn has
00:33:24.982 : been triggered	false	false	false
00:33:24.983 : off time	0	0	0
00:33:24.987 : blank time	16	16	16
00:33:24.987 : hysterisis
00:33:24.987 : -end	-3	-3	-3
00:33:24.987 : -start	1	1	1
00:33:24.987 : Stallguard thrs	0	0	0
00:33:24.991 : DRVSTATUS	X	Y	Z
00:33:24.991 : stallguard
00:33:24.991 : sg_result	0	0	0
00:33:24.991 : fsactive
00:33:24.991 : stst
00:33:24.991 : olb
00:33:24.991 : ola
00:33:24.991 : s2gb
00:33:24.995 : s2ga
00:33:24.995 : otpw
00:33:24.995 : ot
00:33:24.995 : Driver registers:
00:33:24.995 : X = 0x00:00:00:00
00:33:24.995 : Y = 0x00:00:00:00
00:33:24.998 : Z = 0x00:00:00:00

I attached the config files

[code.zip](https://github.com/MarlinFirmware/Marlin/files/1901839/code.zip)

)

teemuatlut, 2018-04-12:
We've got some debugging steps for you to follow on the [Marlin homepage](http://marlinfw.org/docs/features/tmc_drivers.html). I suggest you start with wiring and pin definitions.

* Test the current bugfix branch of Marlin posted on GitHub (in case your issue is already fixed).
* Use the latest TMC Stepper libraries.
* Check all wiring and wire crimps.
* Check 12V (24V) power in the Vm pin and 5V (3.3V) in the Vio pin.
* Check that configured pins match your firmware configuration.
* Enable TMC_DEBUG and send M122 to see the debugging output.
  * Reported register values of either 0x00000000 or 0xFFFFFFFF are bad responses.
* Try the examples provided by the respective library. Please detach any belts beforehand however, as the examples will not respect any endstop signals or physical limits. You may need to change the pin definitions.

If nothing works, please also post a picture of your wiring.

craitor, 2018-04-17:
A baud rate of 200000 isn't a standard that can be selected by repetier however. Howe would I go around this issue?

craitor, 2018-04-18:
Yes, I accidentally looked at a different issue.

So I did some further testing on the hardware, which doesn't seem to be the issue. I replaced the TMC 2130 drivers with regular Polulu drivers and the motors worked fine, which seems to indicate that the Ramps and arduino hardware are intact. 

I checked my pin selections - they seem to be fine.

Here is the wiring:

[image]
[image]

The driver register values are still 0x00:00:00:00.

The config files are identical

craitor, 2018-04-18:
I know there are some other tmc 2130 stepper driver that need to bridge 2 connections on the side of the chip. I don't think these Watterrot stepper drivers do however to enable SPI mode. Am I wrong?

thinkyhead, 2018-04-20:
Good question. I would have to hunt and check the datasheet to answer.

CHILLYSMOKES, 2018-04-21:
You don't need to bridge anything on the Watterotts, only the chinesium ones. Do you use a relay or atx function to remote power on your printer?

landodragon141, 2018-04-27:
Did you solder the pins for SDO, SDI, SCLK, and CS facing down? If so follow Ludy87's advice and pull the config jumpers from under the driver.

Dark-Noir, 2018-06-03:
Similar problem here, everything is setup correctly but after one minute of printing i got a driver error from the extruder driver. Now the drivers are no longer working (all four of them), i tryed reflashing, resoldering etc.
12v and 5v are there, now jumpers set under the drivers.

Running on Marlin bugfix from today.

landodragon141, 2018-06-04:
First off drop your current. You are probably overheating your drivers at 1A. I'd set it to 850-800 and try again. You really don't need that much current for the TMC2130s.

Dark-Noir, 2018-06-04:
> First off drop your current. You are probably overheating your drivers at 1A. I'd set it to 850-800 and try again. You really don't need that much current for the TMC2130s.

Didn't changed anything. I had increased the current because of skipping steps/layer shifting. Also my drivers didn't get hot.

Ludy87, 2018-06-04:
check your Wires, is your SPI connected with SD Card or Display? Try SPI SW on a free pins.
Post your config File.

Dark-Noir, 2018-06-04:
> check your Wires, is your SPI connected with SD Card or Display? Try SPI SW on a free pins.
Post your config File.

Yeah, i have a 12864 display with SD card reader and a BLTouch, so i had to use some other pins. I don't use sensorless homing and i didn't connect the cables for it.

[image]

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2068094/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2068095/Configuration_adv.txt)
[pins_RAMPS.txt](https://github.com/MarlinFirmware/Marlin/files/2068096/pins_RAMPS.txt)

Ludy87, 2018-06-04:
You must change the wire from driver, and enabler 

   #define TMC_USE_SW_SPI
   #define TMC_SW_MOSI       -1
   #define TMC_SW_MISO       -1
   #define TMC_SW_SCK        -1

and write your now pin Numbers

Dark-Noir, 2018-06-04:
> You must change the wire from driver, and enabler
> 
> #define TMC_USE_SW_SPI
> #define TMC_SW_MOSI -1
> #define TMC_SW_MISO -1
> #define TMC_SW_SCK -1
> 
> and write your now pin Numbers

Why i can't use HW SPI?

Ludy87, 2018-06-04:
Try it or forget it😊

Dark-Noir, 2018-06-04:
> Try it or forget it😊

A new Arduino didnt fixed the Problem.
Can i use 63,65,66 from AUX2?

Ludy87, 2018-06-05:
The load on the pins is too high.

landodragon141, 2018-06-09:
If you have a screen and or SD Card attached to the HW SPI it's too much to also put the 2130's on that same bus.

teemuatlut, 2018-06-09:
> If you have a screen and or SD Card attached to the HW SPI it's too much to also put the 2130's on that same bus.

Not true. The TMC2130 only need SPI communication to initialize the parameters when you start the machine. Any comms after that is either because `MONITOR_DRIVER_STATUS` is enabled or you sent a related `M` command.

thinkyhead, 2018-06-10:
@teemuatlut — Any chance that a noisy CS pin could cause data to get bound for the LCD to get to the TMCs and cause mayhem?

teemuatlut, 2018-06-10:
Unlikely as the datagram would have to follow a very specific pattern required by the driver.

towe96, 2018-12-22:
Did software SPI fix your issues? I have the same driver issue - suddenly stopping amidst a print, but software SPI made no difference to me. @Dark-Noir

Dark-Noir, 2018-12-23:
@towe96 I bought a Rumba board, since then no problems.

boelle, 2019-02-20:
@craitor still having an issue?

closed 2019-07-17 by shitcreek