MarlinFirmware/Marlin #10383: Extruder Stepper not working with Repetier

craitor, 2018-04-12:
My extruder motor doesn't move.

I have tested the motor on a different axis. - the stepper motor works
Also I have tested the stepper driver of the extruder on a different axis. - the stepper driver works
The issue seems to be software related.

I am running Marlin 1.1.8 and using the Repetier software.

I added the code below:
[troubleshoot.zip](https://github.com/MarlinFirmware/Marlin/files/1901870/troubleshoot.zip)

craitor, 2018-04-12:
I don't get any error code in Repetier. Only the motor doesn't get power.

ManuelMcLure, 2018-04-12:
This could be caused by `PREVENT_COLD_EXTRUSION` if your hot end temperature is below 20C (per your configuration file).

thinkyhead, 2018-04-12:
Try `M302 S0` and see if it solves the issue. If that doesn't allow "cold extrusion" then…

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

closed 2019-05-16 by thinkyhead