MarlinFirmware/Marlin #10384: No Extrusion if retract value over 4mm

phillipsko, 2018-04-12:
### Description
No Extrusion if retract value over 4mm
Marlin 1.1.8rc

### Steps to Reproduce
1. retract value of 4mm or more 
2. slice
3. start job with octoprint

**Expected behavior:** 
print should work

**Actual behavior:**
start gcode inkluding nozzle prime ,working as expected
if the actual slicer gcod starts, extruder stops moving, no extrusion at all
Tested with Slic3r pursa edition, cura 3.2.1, S3D 4.0

thinkyhead, 2018-04-13:
Firmware retraction or slicer-based retraction?

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

boelle, 2019-02-20:
@phillipsko tried what thinky suggested?

phillipsko, 2019-02-20:
Uff that's a long time ago I really don't know what I did to get it back
working, meanwhile i changed to Klipper fw.

boelle, 2019-02-20:
@phillipsko then maybe hit the green close botton below?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-11 by shitcreek