MarlinFirmware/Marlin #10388: M118 Not working as expected

benlye, 2018-04-12:
### Description

The `M118` command is not working as expected when trying to send output to the console. 

1. When the text includes an `A` or an `E` anywhere in the string the output misbehaves
1. The entire string gets converted to uppercase

I've tested in bugfix-1.1.x and bugfix-2.0.x and found the same behaviour in both.

### Steps to Reproduce

#### No parameter

**Command:** 
`M118 A bad thing happened!`

**Expected behavior:**
```
Send: M118 A bad thing happened!
Recv: A bad thing happened!
Recv: ok
```

**Actual behavior:**
```
Send: M118 A BAD THING HAPPENED!
Recv:  BAD THING HAPPENED!
Recv: ok
```

####  `A` parameter
`M118 A1 ACTION:dostuff`

**Expected behavior:**
```
Send: M118 A1 ACTION:dostuff
Recv: //ACTION:dostuff
Recv: ok
```

**Actual behavior:**
```
Send: M118 A1 ACTION:dostuff
Recv: ACTION:dostuff
Recv: ok
```

3. `M118 E1 Message goes here`

**Expected behavior:**
```
Send: M118 E1 Message goes here
Recv: echo:Message goes here
Recv: ok
```

**Actual behavior:**
```
Send: M118 E1 MESSAGE GOES HERE
Recv: MESSAGE GOES HERE
Recv: ok
```

### More information
It seems that if an `A1` parameter is given then the presence of a letter `A` anywhere in the string breaks the expected behaviour.  If an `E1` parameter is given the presence of the letter `E` anywhere in the string breaks.

Also, the `A1` behaviour as implemented is broken as it puts a space after the `//`, when OctoPrint expects no space between the slashes and the commands.

It seems like this is just trying to be too clever, and adding unneeded complexity.  Could we scrap the `A` and `E` parameters altogether and just make it behave like `M117` - the whole argument is treated as a string and sent to the console as entered.  That would leave it up to the user to prepend `//`  or `echo:` as they need.

benlye, 2018-04-12:
My testing of this behaviour was partially thwarted by OctoPrint, which converts terminal commands to uppercase for all commands except `M117` by default.  I've opened a PR on OctoPrint (https://github.com/foosel/OctoPrint/pull/2573) to add `M118` to the auto-uppercasing blacklist.

thinkyhead, 2018-04-13:
Now patched in the bugfix branches. Please confirm whether it works and close this issue if we got it.

benlye, 2018-04-13:
Perfect - thanks!

commit 2edc13a0 (2018-04-13): Fix M118 parameter handling

Fix #10388
commit 4c875172 (2018-04-13): Fix M118 parameter handling

Fix #10388
closed 2018-04-13 by benlye