MarlinFirmware/Marlin #10390: Mesh bed leveling doesn't handle Z_MAX homing properly

bryanboettcher, 2018-04-12:
### Description

I use a MakerGear M2 which homes to Z_MAX.  When starting mesh bed leveling with manual probing (the paper-drag test), G29 S1 homes all 3 axises leaves the bed at Z_MAX but behaves as though the bed is near the nozzle.

### Steps to Reproduce

Configure a machine with the following settings:

* MESH_BED_LEVELING
* PROBE_MANUALLY
* Z_HOME_DIR 1
* Z_MAX_POS 205 (or whatever)
* MIN/MAX_SOFTWARE_ENDSTOPS disabled
* LCD_BED_LEVELING

Operate the LCD menu command for *Prepare > Bed Leveling > Level Bed*

**Expected behavior:** 
The machine will home all 3 axises, move to probe point 1, then move the bed into place to allow for Z adjustment.

**Actual behavior:** [What actually happens]

The machines homes all 3 axises, moves to probe point 1, leaves the bed at Z_MAX, and presents the menu to allow fine-tuning of the Z position.

Roxy-3D, 2018-04-12:
> I use a MakerGear M2 which homes to Z_MAX.

> MIN/MAX_SOFTWARE_ENDSTOPS disabled

So you don't have a Z-Max endstop defined and working????

thinkyhead, 2018-04-12:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

bryanboettcher, 2018-04-13:
This was with bugfix-1.1.x as of commit 21971f2f7a64f43c333a6a8c3d5471de43b5670f

thinkyhead, 2018-04-13:
Incidentally, what is your `MANUAL_PROBE_HEIGHT` setting?

thinkyhead, 2018-04-13:
I've patched the bugfix branches. Please check to confirm whether it's fixed, and if so then we can close this issue.

bryanboettcher, 2018-04-13:
Z_CLEARANCE_BETWEEN_PROBES is the default 5, and Z_HOMING_HEIGHT is undefined.

After your changes, the homing sequence for bed leveling went to 205 to hit the endstop, then up to the nozzle (0 / Z_MIN_POS), then back down to 205, then up to Z_MIN_POS again.

thinkyhead, 2018-04-13:
Interesting. I was expecting this sequence:
- Home Z to max (205).
- Move to XY of the first point to probe.
- Move Z to 0 (close to the bed).
- Print "`MBL G29 point 1 of X`" to serial.
- Print current position to serial.
- Idle and wait for Z adjustment.

I'm not sure how it's going back up to Z=205 since `G29 S2` is only done once per point, and the first one modifies `current_position`. I'll keep looking into it.

Do you have enough space to use `AUTO_BED_LEVELING_BILINEAR` with `PROBE_MANUALLY`? If so, test that to see if that behaves any better. We might have to kill two birds with one stone.

Be sure to download the current `bugfix-1.1.x` since it's had some tweaks here and there in the last hour or so.

bryanboettcher, 2018-04-13:
ABL_BILINEAR does the same "bed doesn't go to 0" behavior that MESH did originally.  I have a print running on it now, but I don't remember it zeroing twice when I switched back to MESH.

thinkyhead, 2018-04-14:
I did some small tweaks to `MESH_BED_LEVELING` as part of the last update. If that's working now, great. I'll have to make changes to ABL_BILINEAR to match.

bryanboettcher, 2018-04-14:
Should this be respecting the home offsets for Z?  I have a 0.5mm offset in Z, and my entire bed mesh is about 0.5mm off -- as though the offset wasn't applied (my bed mesh goes from 0.47 - 0.65).  When the print starts, the bed is too close by about the same amount, like it's now double-applying the offset (once from M206, once from G29).

Am I supposed to be using a different mechanism for offset, now?

thinkyhead, 2018-04-16:
The purpose of the `M206 Z` home offset is to correct for a deviated Z endstop on `G28` so that when you do `G1 Z0` your nozzle should be just touching the bed. 

Later when you do `G29`, wherever the bed is at the "perfect" height, your mesh values will be 0.

bryanboettcher, 2018-04-16:
That was my assumption as well.  What seems to be happening though is the G28 before homing isn't applying the home_offset[Z_AXIS] before moving the nozzle into Z_MIN_POS.  When the nozzle approaches the bed, it's about my home_offset distance away, and then the entire mesh is about home_offset off.

brookstalley, 2018-04-17:
This sounds similar to what I experienced with a MakerGear M2: I couldn't get any permutation of M851 and M206 to produce satisfactory results. Discussion in [this thread](https://github.com/MarlinFirmware/Marlin/issues/9418#issuecomment-363150989).

I ended up moving my M206 value into the compile-time Z_MAX_POS so it is included in the machine coordinate space. Where I used to have Z_MAX_POS 200 and M206 Z9.54, I now have Z_MAX_POS 209.54 and M206 0. That's simplified life and UBL works as expected, producing small numbers on either side of 0.

thinkyhead, 2018-04-18:
@bryanboettcher — What is your `Z_MIN_POS` value?

I'm thinking I should change all moves "close to the bed" to just use 0.0, since that should universally be the height of the bed (in terms of native machine position).

My use of `Z_MIN_POS` for these moves is clearly an error on my part, since the purpose of that setting (as pertains to Z) is almost always to give some extra range of motion below the bed and set that as the default Z min software endstop.

`Z_MIN_POS` also doubles as the Z home position when `MANUAL_Z_HOME_POS` isn't set, mostly for parity with XY, where it's a safer assumption that the movement limits correspond to the endstop positions. Almost all the time, what you really want for `Z_HOME_POS` is 0.

thinkyhead, 2018-04-18:
Ok, so I've patched the `bugfix-1.1.x` branch with those considerations in mind. Please give it a try to see if it has regained some sanity.

bryanboettcher, 2018-04-18:
I haven't tried the new firmware yet, I will tomorrow, but some other
thoughts...

My Z_MIN is set to 0. My Z_MAX is set to 205, because that's a good average
across the 6 M2s I'm running. I tweak the home position with M206, and
that's the MakerGear method as well.

It seems like if the homing code applies home_offset right after the
endstops were hit, the problem would be resolved. For example, on M2 #3 for
me, it has a 2.2mm offset on Z. When I run "G28 Z0", followed by an M114, I
get Z205 instead of the Z207.2 I'd expect.

I need to double check that behavior, and probably open a second bug, but
that's how G29 is behaving for me.

In the meantime, I can work around it with the method above.

thinkyhead, 2018-04-18:
> It seems like if the homing code applies `home_offset` right after the endstops were hit, the problem would be resolved.

That's how it's supposed to work. What is the current position according to `M114` immediately after a `G28`?

bryanboettcher, 2018-04-18:
To test the behavior and really try and narrow it down, I pulled latest from bugfix-1.1.x and really exaggerated Z_MAX_POS.

I set Z_MAX_POS to 100, and used M206 to set a Z offset of 105.  This would make an effective height of 205, which is close enough to what it should be for illustration.  G28 Z0 -> G1 Z0 brought the nozzle straight to the bed, as expected.  However, if I go to MBL, I get this:

```
SENT: G29 S1
READ: enqueueing "G28"
READ: MBL G29 point 0 of 12
READ: X:20.00 Y:20.00 Z:105.00 E:0.00 Count X:1778 Y:1778 Z:0
READ: ok
READ: enqueueing "G29 S2"
READ: X:0.00 Y:0.00 Z:205.00 E:0.00 Count X:0 Y:0 Z:40000
READ: busy: processing
READ: MBL G29 point 1 of 12
READ: X:20.00 Y:20.00 Z:105.00 E:0.00 Count X:1778 Y:1778 Z:0
```

The bed is squarely in the middle of the printer and not at the nozzle as it was from G28 -> G1 Z0.

Roxy-3D, 2018-04-18:
> That's how it's supposed to work. What is the current position according to M114 immediately after a G28?

And it probably would be helpful to build with #define DEBUG_LEVELING_FEATURE enabled...  and do a M111 S255 prior to doing the G28.

bryanboettcher, 2018-04-18:
Will do, Roxy-3D.  It's worth noting that G28 -> M114 reports the correct Z205 (MAX_POS 100 + Z_offset 105).

thinkyhead, 2018-04-20:
I see what's going on.

If you tell Marlin that your machine is 100mm tall, it will take your word for it and that will apply in many places. The software endstops will remain 100mm apart, no matter what.

`M206` doesn't just tell Marlin what value to add to the current position when homing. It also offsets the logical coordinate space.

With your machine height (`Z_MAX_POS`) set to 100, `M206 Z105` makes your logical home position Z=205, and also moves the min software endstop to the logical position Z=105. Internally, when a move is done to Z=0 (machine space) this adds 105, so the bed ends up 105mm away from the nozzle.

Since a move like `G1 Z0` subtracts the home offset for Z, the nozzle will move all the way down to the machine space position Z=-105 (if software endstops are disabled). With software endstops enabled, a `G1 Z0` move will stop when the machine reaches the place where it thinks the bed is located, which is Z=105.
```
machine space          logical space

            100 ---- 205 ------- max soft endstop, z home pos
             ^        ^
             |        |
             |        |
             v        v
             0  ---- 105 ------- min soft endstop, bed position
             ^        ^
             |        |
             |        |
             v        v
           -105 ----  0
```

thinkyhead, 2018-04-20:
I see what's going on.

If you tell Marlin that your machine is 100mm tall, it will take your word for it and that will apply in many places. The software endstops will remain 100mm apart, no matter what, and the machine will be defined as 100mm tall, no matter what. The bed will always be 100mm away from the max endstop, even after the endstop's logical coordinate is changed with `M206 Z`.

`M206` doesn't just tell Marlin what value to add to the current position when homing. It also offsets the logical coordinate space.

With your machine height (`Z_MAX_POS`) set to 100, `M206 Z105` makes your logical home position Z=205, and also moves the min software endstop to the logical position Z=105. Internally, when a move is done to Z=0 (machine space) this adds 105 (from the logical perspective), so the bed (machine space Z=0) is always 100mm away from the Z endstop.

Since a move like `G1 Z0` subtracts the home offset for Z, the nozzle will move all the way down to the machine space position Z=-105 (if software endstops are disabled). With software endstops enabled, a `G1 Z0` move will stop when the machine reaches the place where it thinks the bed is located, which is Z=105.
```
machine space          logical space

            100 ---- 205 ------- max soft endstop, z home pos
             ^        ^
             |        |
             |        |
             v        v
             0  ---- 105 ------- min soft endstop, bed position
             ^        ^
             |        |
             |        |
             v        v
           -105 ----  0
```
So your assessment is correct. Marlin doesn't expand the size of the machine or move the bed based on the `M206 Z` home offset. It simply adjusts where Z=0 is located relative to the Z endstop.

bryanboettcher, 2018-04-20:
Should I open a new issue for this then?  Maybe I'm misunderstanding how I should be using some features.  The original one is properly fixed :)

brookstalley, 2018-04-20:
Great explanation, @thinkyhead . 

If I'm interpreting correctly, on a machine which homes to z-max and which uses a probe to set z=0, there is an overconstraint where Z_MAX_POS specifies the height of the machine, but the actual height is actually variable depending on distance between probe activation and endstop (SW or HW).

So changing the bed height (glass to MIC6, for instance), means that Z_MAX_POS has to be adjusted if the machine space numbers are going to work out (which is required for successful bed leveling, since it operates in machine space). 

If that's correct so far, does this look like the right sequence to adjust values:
- Set Z_MAX_POS to an approximate value for machine height
- G28 (machine goes to z-max endstop and M114 reports the Z_MAX height)
- Bring bed up to true 0 position, just touching nozzle
- M114 to get Z pos relative to z-max endstop
- Subtract reported value from Z_MAX_POS (if negative, Z_MAX_POS was too low)

thinkyhead, 2018-04-21:
Essentially correct, @brookstalley …

The issue is in how we have conceived the `M206` home offset. If it wasn't tied to the machine space offset, it might be better. There's nothing absolutely requiring it to be so. It could simply get added to the current position on `G28` (as it was in Marlin 1.0.0) and then have no further effect.

However, this still brings it into potential conflict with soft endstops and the machine size. For example, if you were homing to min, typically you want your `Z_MIN_POS` to be 0 and your Z min endstop to be "perfect" so that the nozzle is touching the bed with `G1 Z0`. If you've got a loose or adjustable bed, then you might use `M206 Z-1.5` to indicate that the bed is higher than the endstop. With min soft endstops this leads to all moves below the bed being constrained to Z=0 (i.e., the configured `Z_MIN_POS` value). And most of the time that's what you really want. You don't generally want to set the bed as being _lower_ than the Z min endstop, because you end up mashing the Z endstop on the lower layers. And, if min endstops are enabled then the nozzle won't be allowed to go lower than the machine space Z — I.e., all `G1` moves will be constrained to a height above the bed.

I've been back and forth about how best to implement `M206`, how it should affect the software endstops, and how it affects the logical coordinate space. And with this issue I've had to re-think it again. At one point I had it so the home offsets would expand the soft endstops outward (if negative on min or positive on max), but they would not bring them inward (if positive on min or negative on max). Each time I've made a tweak someone has expressed caveats and I've had to reexamine them again.

So, the basic considerations are:
- Home offsets should obviously affect the current position upon homing.
- The full movement range of the machine (configured distance from `Z_MIN_POS` to `Z_MAX_POS`) has to be preserved.
- The software endstops need to correlate with the full movement range (since they enforce it).
- Commands like `G1` need to arrive at the expected logical position. For example, if your `X_MIN_POS` (and X home pos) are -10 and you set the X home offset to -1, making the X position -11 on `G28`, then `G1 X-11` should move to the home position. (Currently this works as a consequence of `M206` applying to the workspace offset.)

Some of the behavior described in this issue can be mitigated by setting up all the machine configurations so the `Z_MAX_POS` is always bigger than needed, and never smaller. They will start out having the nozzle too far from the bed until `M206` fixes things. Then those that need adjustment can have the appropriate negative home offset applied. The only extra side-effect will be that the Z min soft endstop will allow movement below the bed.

That side-effect could be mitigated by always forcing the Z min endstop to always correlate with the logical Z=0 position, which 99.89% of the time is what you would want anyways. (There are not a lot of machines where it's desirable or necessary to be able to move the nozzle below the bed.)

Note that the behavior of bed leveling is not hampered by the soft endstops — they're only applied to the logical coordinates in commands like `G1` and not to all motion generally.

brookstalley, 2018-04-21:
Thanks for the really thorough analysis. I hear you that almost any solution is going to cause problems for some permutation of machine config.

What if we step back and say that each axis must have exactly two end stops: each one can be either SW or HW.

In the case of our machines that home to zmax and have a probe or manual adjustment to set zmin =0, there is no room for another variable specifying machine height. You set the bed to z=0, you home with G28, and M114 tells you Z_MAX_POS.

Practically, I think that means an asset on compile that breaks if Z_MAX_POS is specified in addition to both a max and min sensor (probe). 

If an axis has one mechanical end stop, you can/must set the other side, whether it’s firmware config like X_MAX_POS or M206.

That seems like a way to ensure machine space min/max are set but not overconstrained. I think the heart of the problem is having three specifiers for how long a line is.

From there to logical space... I will admit I don’t have it.

thinkyhead, 2018-04-22:
> What if we step back and say that each axis must have exactly two end stops: each one can be either SW or HW.

So-called "software endstops" are just there to prevent movement beyond the defined physical bounds, so they aren't really "endstops" as much as they are "limits." Yes, endstops are also sometimes called "limit switches," but really their job is not to enforce limits but to establish a known (e.g., "home") position.

Anyway, I think it's good to have both "movement limits" and "position-switches" and not to conflate the two features.

At the moment, these limits are established at boot and they don't change (except automatically, for dual X carriage). It seems that they should change under some other conditions too, such as when you've switched to a different nozzle. From this issue, I think it also makes sense to change these limits when the home offsets change, so that expansion of the space is properly considered.

I'll work on this over the next day or so and see that it works in a way that this issue gets solved.

bryanboettcher, 2018-04-25:
Is this a thing I can use "Bed Z" for?

Also, you may want to do this instead -- my Z move was hitting the binder clips:
`          do_blocking_move_to_xy(mbl.index_to_xpos[0], mbl.index_to_ypos[0]);
          do_blocking_move_to_z(0);
`

vjvarada, 2018-05-08:
If you set the Z home offset to 0, do the mesh bed leveling, and then set the Z home offset back, would the Mesh be offset by the appropriate amount?

thinkyhead, 2018-05-10:
> If you set the Z home offset to 0, do the mesh bed leveling, and then set the Z home offset back, would the Mesh be offset by the appropriate amount?

The mesh itself is not altered, but `G29 S4 Znnn` changes the MBL Z offset, which is used throughout leveling compensation. When changed it takes effect on the next move.

bryanboettcher, 2018-05-10:
Is perhaps Bed Z (which is what it's called in the LCD menu) what I want
instead of home offset?  The problem I was having with the entire mesh
being -1.5ish mm is the parts ended up 1.5mm too tall because of the
fading.  If Bed Z is set to -1.5, Home Offset to 0, and levelled then --
will my parts end up the correct height?

thinkyhead, 2018-05-10:
If your mesh has values tending towards 1.5mm in either direction, that's primarily due to not having a good home position, such that Z0 puts the nozzle right on the bed. 

The MBL `z_offset` is not added to the mesh value before applying fade, so `G29 S4 Z` alone won't help. The calculation is `z_offset + z0 * fade`, not `(z_offset + z0) * fade`.

With the latest bugfix code, you can now use `M420 C` to center the mesh values around 0. If after making that adjustment your Z0 is still somewhere off the bed, `G29 S4 Znnn` _can_ be used to move the nozzle closer to the bed.

commit 942a4072 (2018-04-13): Make sure MBL starts close to the bed

Fix #10390
commit dc549095 (2018-04-13): Make sure MBL starts close to the bed

Fix #10390
commit daa71245 (2018-04-13): Make sure MBL starts close to the bed

Fix #10390
commit db81a1d4 (2018-04-13): Make sure MBL starts close to the bed

Fix #10390
closed 2018-07-31 by thinkyhead