MarlinFirmware/Marlin #10394: how to add a phusical button to extruder motor control?

amrithmmh, 2018-04-13:
i am trying to add a physical button to control the stepper motor (extruder) so i can just extrude or reverse it by pressing the button easily other than the lcd control.
Can i add a button(or two) as input to arduino and add my code in stepper.cpp?
which pin(remaining after ramps 1.4) can i use to achieve this?

all suggestions are  welcome

TheSFReader, 2018-04-13:
rather than directly work on stepper.cpp, I would rather suggest to take a look at the LCD control part, and integrate an emulation of the prepare/move/Z-axis code. As for accessible pins, perhaps use any unused endstop ones ?

boelle, 2019-02-20:
@amrithmmh 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

amrithmmh, 2019-03-02:
@boelle its a feature request rather than a question

closed 2019-03-02 by amrithmmh

thinkyhead, 2019-03-03:
This is not a feature we would add. But it's possible to do movement with buttons by hooking into the LCD code and following the technique currently used for the KEYPADs.