MarlinFirmware/Marlin #10395: 2.0.x - 13 April - u8g.h bug (again)

Nume1977, 2018-04-13:
This seems to be a reversion bug, tested with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER enabled:

```
sketch\src\lcd\u8g_fontutf8.h:12:17: fatal error: u8g.h: No such file or directory

 #include "u8g.h"
```
or
```
sketch\src\lcd\u8g_fontutf8.h:12:17: fatal error: u8g.h: No such file or directory

 #include <u8g.h>
```

rmoravcik, 2018-04-13:
Try to replace u8g.h with clib/u8g.h, this worked for me

thinkyhead, 2018-04-14:
Try `#include <U8glib.h>`.

thinkyhead, 2018-04-14:
@yhfudev — We'll need a definitive fix for this issue.

thinkyhead, 2018-04-14:
@yhfudev — We also need good documentation on how to use the scripts in `buildroot/share/fonts`. I have some patches for items in that folder which I will post shortly in another PR for your review. Meanwhile, please review a92776d.

p3p, 2018-04-14:
> I'm not sure where "u8g.h" is supposed to be, but currently not all user build tools are finding it.

It's in the lib folder in Marlins u8glib-HAL library, clib in the normal Arduino version, I'm not sure how Arduino decides what sub directories to add the the Include Path or why it stopped working, neither library specifies in the associated library descriptor. No idea why it stopped being found..

yhfudev, 2018-04-14:
The current source code can't support both of "end user" and "travis build system" at the same time. The path 'lib/u8g.h' (u8glib-HAL) currently only used in the travis/platformio script. so we need to change the include to '<clib/u8g.h>'. Fortunately, the clib/u8g.h needed only in one file u8g_fontutf8.h. I have working on this issue, and will submit the PR with a set of changed files that can support both "end user" and "travis".

p3p, 2018-04-14:
I presume if it really is necessary to bypass the api and access the c header directly from the subdirectory the directory name can be changed in Marlins u8glib-HAL to simplify things.

yhfudev, 2018-04-15:
It is needed by the utf8 engine in the file u8g_fontutf8.c,
The PR #10419 will handle the "lib/u8g.h" and "clib/u8g.h" problem.

p3p, 2018-04-15:
We need to fix this without workarounds, such as the added Include path in platformio.ini, or the Travis only fix in your PR, normal PlatformIO builds will still be broken.

commit 58e077db (2018-04-14): u8g.h => U8glib.h

Fix #10395
Followup to #10213
commit 7c68ebfb (2018-04-14): u8g.h => U8glib.h

Fix #10395
Followup to #10213
commit a92776d1 (2018-04-14): u8g.h => U8glib.h

Fix #10395
Followup to #10213
commit e0cb8ff7 (2018-04-14): u8g.h => U8glib.h

Fix #10395
Followup to #10213
closed 2018-04-15 by thinkyhead