MarlinFirmware/Marlin #10396: [1.1.x-bugfix] "Stop print" during G28 causes nozzle to crash into bed

gloomyandy, 2018-04-13:
### Description
Using the LCD "Stop program" option to stop a program that is performing a G28 operation can result in the nozzle crashing into the bed.

### Steps to Reproduce

1. Run a program from the SD card that contains a G28 command
2. Wait until the G28 is executed and the printer has homed first X and then Y. Then choose "Stop print".
**Expected behavior:** [What you expect to happen]
Everything should stop.

**Actual behavior:** [What actually happens]
G28 continues, but does not move the probe into the centre of the bed (as it normally would), as a result the probe is off the edge of the bed (with my configuration) and so does not detect it and the nozzle crashes into the bed.

#### Additional Information

Configuration files here:
https://github.com/gloomyandy/Marlin/tree/bugfix-1.1.x

Note that reproducing this may be timing dependant you need to cancel the print just as the printer has completed X and Y home (or possibly just before it completes Y home). Due to the damage it can do I've not tried to reproduce this, nor do I really want to!

thinkyhead, 2018-04-14:
You think that's bad, you should see what happens if you stop `G29` in-progress.

The easiest solution is to have "Stop SD Print" reboot the printer. Other solutions tend to get complicated.

thinkyhead, 2018-04-14:
The simplest way to abort a print is to allow a lengthy command to complete. That's what the PR linked above does.

gloomyandy, 2018-04-14:
Thanks for taking a look at this. Makes a lot of sense not to attempt an uncontrolled abort of complex commands (which could have all sorts of bad consequences), but rather to allow them to decide how to be aborted (if at all), safely.

marcio-ao, 2018-10-19:
In our printer, I implemented a fairly complicated calibration routine as a GCODE. I've gotten one complaint about this being non-interruptible as well. So I'll chime in and say that I think it would be awesome if Marlin had a global flag that would mean "abort current operation as soon as possible". It would be up to individual GCODEs to poll that flag, but I think if it were provided, it would get used.

thinkyhead, 2018-10-26:
M410: Quickstop - Abort all planned moves
M112: Emergency Stop

boelle, 2019-02-20:
@gloomyandy could this one be closed or might there be a purpose hidden from my wisdom? :-D

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead