MarlinFirmware/Marlin #10397: Problem with BLTOUCH and G28 command

BEUSX, 2018-04-13:
hi, I need your help.
I have an i3 Mega anycubic Which has two endstops on the Z axis.I modified the firmware to be able to use BLtouch...and work fine! but when i do the command G28 , the sensor deploy the pin....so i can't use the 2 endstops to do the "home" and level the Z axis.
There is a way to prevent the Bltouch from being active when running g28? (not deploy the pin)
i disable the "Z_SAFE_HOMING", but don't work...it deploy.
i need that the sensor not deploy the pin when i do a g28 command...i want to use it only for bed probe.

sorry for my english .....I recently started studying the Marlin firmware. i'm not good....i know.....
tnx!

BEUSX, 2018-04-14:
OK TNX!!! I TRY!!!!!!!!!!!

boelle, 2019-02-20:
@BEUSX did you try it?

closed 2019-07-11 by shitcreek