MarlinFirmware/Marlin #10398: [1.1.x-bugfix] UBL G29 J3 does not stow the probe and can cause head to crash into frame

gloomyandy, 2018-04-13:
### Description
Using the UBL G29 J3 option to perform a 3x3 grid update of the UBL mesh rather than the normal G29 J 3 point adjustment will leave the probe not stowed. This also seems to have the effect of leaving the end stops enabled which in my case can cause a print using the G29 J3 command to end up crashing the head into the frame. 

### Steps to Reproduce
1. Print a file that contains a sequence like...
G28 
G29 L1
G29 J3
G29 A

**Expected behavior:** [What you expect to happen]
System should probe a 3x3 grid to use to adjust the UBL mesh. The probe should be stowed and the print should continue with the corrected mesh.

**Actual behavior:** [What actually happens]
System performs the 3x3 probe, but does not stow the probe. This seems to leave end stops enabled and a number of reports of "echo:endstops hit" can be seen on an attached console. This includes reports of Z end stop being constantly triggered (because in my case it is a probe, close to the bed). These spurious triggers seems to disrupt the print motion eventually causing the head to crash into the frame.

#### Additional Information

Configuration files here:
https://github.com/gloomyandy/Marlin/tree/bugfix-1.1.x

Suggested fix is to add a new line to tilt_mesh_based_on_probed_grid to call STOW_PROBE() after the end of the probing loop. See:
https://github.com/gloomyandy/Marlin/commit/3e62f36389d880adc880193d961737e1eed4622c
The additional change here adds verbose output to bring G29 J3 more inline with G29 J.

Not sure if the end stops should be being left enabled after calls to probe_pt or not, but they certainly seem to be if PROBE_PT_RAISE is used rather then PROBE_PT_STOW.

Roxy-3D, 2018-04-13:
I'll try to duplicate this later today.   If I can duplicate it,  it will be easy to put your changes in and verify correct operation.

THANKS for the fix!

gloomyandy, 2018-04-13:
I'll update and check them things tomorrow and then close this if all is well!