MarlinFirmware/Marlin #10399: I2C in Arduino for Marlin

kcheeeung, 2018-04-13:
Hey guys! I would like to understand using I2C Arduino for Marlin. I'm planning to use the SDA/SCL pins to communicate with a few peripherals mainly to extend the board's capabilities. Thanks for reading.

1. Is it computationally "intensive" to keep the I2C active and send commands to slave peripherals? I believe that for LCDs, the board continuously communicates to the slave. I have an ATmega2560.
2. Is the Marlin using the Wire lib provided by Arduino or a custom library for I2C? And possibly point where I can look into that?
3. Is it possible to transmit data to the slave device "on-demand?" 
I'm thinking of putting the transmission protocol in the gcode (ex. start to communicate with XXX address and do XXX) so the board will communicate with the slave whenever it sees that particular gcode. So I will just keep the SDA/SCL lines active with `Wire.begin()` in the void setup (if using the Arduino's Wire library).

thinkyhead, 2018-04-14:
1. It's not too intensive. Many peripherals use i2c.
2. It uses `Wire` on most platforms. Enable `EXPERIMENTAL_I2CBUS` for a simplified interface in Marlin, and check out this thread for insights into using i2c to talk to a second board: https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879
3. Yes, you can send data over i2c anytime. See the aforementioned thread for more insights into using the `EXPERIMENTAL_I2CBUS` to test i2c communication.

kcheeeung, 2018-04-17:
Great! Thanks for the clarification!

closed 2018-04-17 by kcheeeung

Molde, 2018-12-29:
Hey there,
i have a short question.

The communication between my ESP 8266 and the Arduino Mega (running Marlin) works.

But I just can't send a simple command to execute a simple movement.

For example:

When I send G01 X10 -> the x motor should move but when i send "G01 X10" via I2C from ESP to Marlin, Marlin only writes: 

i2c-reply: from:99 bytes:5 data: G01 X10

But nothing happens. Anyone knows how to handle the gcode in Marlin to execute the command?

Thank you very much!!! :-)

kcheeeung, 2018-12-29:
@Molde Are you sending your gcommands through a serial port to your printer directly using your wifi board? Or are you somehow connecting to your printer through the I2C lines (SCL and SDA) and then sending that gcommand? I2C uses binaries to communicate back and forth with its peripherals.

miltostsalekkas, 2018-12-30:
Hey, i am facing the same situation.How can we implement the gcode sent by the master board?I am getting the echo on the slave, assuring me that it got the command, but how can we print it?

kcheeeung, 2018-12-31:
@miltostsalekkas 
It sounds like someone has already gotten to your answer in #12756, covering arduino to arduino communications.
How I used it was master (3d printer board) -> slave (peripheral), so some custom gcode will turn on/off my peripheral device. If this covers part of your application, feel free to let me know if you need any help.

miltostsalekkas, 2018-12-31:
Hey. Thank you very much. I already solved it.