MarlinFirmware/Marlin #10400: Marlin 2.0x sbase no sd card

warbunnies, 2018-04-13:
What changes need to be made to the current firmware to get the software sd card to work?

I have:
- enabled in the config file & #define SD_CHECK_AND_RETRY
-broken out the miso, mosi, & sck to the suggested locations and check 4 times that it was wired right.
-also broke out the SD detect pin because it wasn't working for some reason...but now it is on pin 2_13!
-I tried several different spi_pin.h files people have been floating around and nothing worked. Is there a "this spi_pin.h file is the officially right one of the sbase"?

I'm running out of things to try.  Any ideas?

p3p, 2018-04-15:
> I tried several different spi_pin.h files people have been floating around and nothing worked. Is there a "this spi_pin.h file is the officially right one of the sbase"?

As far as I'm aware the officially correct one is the one we have, there should be no need to change it. Hopefully someone with an mks sbase can help you, I maintain the LPC1768 platform but do not have access to this board, only the connector is different though, it 'should' work when wired correctly.

warbunnies, 2018-04-15:
Hmm... maybe something is wrong with my cable then. The sd card detect on the main screen didn't work until I broke out the pin. Thank you for updating me. I'll try some molex connectors and see if i can get this working.

michaelwu21, 2018-04-18:
@warbunnies Have you gotten yours to work? 

I've tried several fixes from past posts but my sd card is still not detected and eeprom is not saved.

warbunnies, 2018-04-19:
I will see if switching the SDss pin from 0_06 to P0_28 or unspecified and see what happens

p3p, 2018-04-19:
The schematic I have shows that as being correct, but I can't confirm it without the hardware. It shouldn't be a problem unless you are using a character display (doesn't use spi).

warbunnies, 2018-04-19:
Seems I've been messing around too much. I got a new error:
arm-none-eabi-objcopy:.pioenvs\LPC1768\firmware.bin: Permission denied
*** [.pioenvs\LPC1768\firmware.bin] Error 1

thoughts?

p3p, 2018-04-19:
something must have it open or didn't release the file properly, suggest a restart

warbunnies, 2018-04-19:
Yep just needed a restart. It probably didn't release when I uploaded it to my email

warbunnies, 2018-04-23:
This might help clear things up for me a little bit. 
I'm using a #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

the pins_mks_sbase.h file only has:
#if ENABLED(**ULTRA_LCD**)
  #define BEEPER_PIN       P1_31   // EXP1.1
  #define BTN_ENC          P1_30   // EXP1.2
  #define BTN_EN1          P3_26   // EXP2.5
  #define BTN_EN2          P3_25   // EXP2.3
  #define SD_DETECT_PIN    P2_13   // EXP2.7
  #define LCD_PINS_RS      P0_16   // EXP1.4
  #define LCD_SDSS         P0_28   // EXP2.4
  #define LCD_PINS_ENABLE  P0_18   // EXP1.3
  #define LCD_PINS_D4      P0_15   // EXP1.5
#endif // ULTRA_LCD

but the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is a DOGLCD 
So where are the pins for the Doglcd defined???

thinkyhead, 2018-04-24:
> The Display/Controller defines are a bit of a maze

To help navigate the maze…

https://docs.google.com/spreadsheets/d/1rV5VhMA743SQAjUMBd6W15QVrLIWKVxvxFSBdhgj6Rk/edit?usp=sharing
[image]

warbunnies, 2018-04-24:
Thank you. What does the varying levels of green mean?

warbunnies, 2018-04-24:
I have been digging around for any sorta answer. The smoothieware defines the spi speed as 500000. I was wondering how to set the spi speed. I've been digging around and found this:

Marlin_speed[0] = 8333333; //(SCR:  2)  desired: 8,000,000  actual: 8,333,333  +4.2%  SPI_FULL_SPEED
    Marlin_speed[1] = 4166667; //(SCR:  5)  desired: 4,000,000  actual: 4,166,667  +4.2%  SPI_HALF_SPEED
    Marlin_speed[2] = 2083333; //(SCR: 11)  desired: 2,000,000  actual: 2,083,333  +4.2% SPI_QUARTER_SPEED
    Marlin_speed[3] = 1000000; //(SCR: 24)  desired: 1,000,000  actual: 1,000,000         SPI_EIGHTH_SPEED
    Marlin_speed[4] =  500000; //(SCR: 49)  desired:   500,000  actual:   500,000         SPI_SPEED_5
    Marlin_speed[5] =  250000; //(SCR: 99)  desired:   250,000  actual:   250,000         SPI_SPEED_6
    Marlin_speed[6] =  125000; //(SCR:199)  desired:   125,000  actual:   125,000         Default from HAL.h

which sounds like it is used when setting:

 * SD CARD: SPI SPEED
 *
 * Enable one of the following items for a slower SPI transfer speed.
 * This may be required to resolve "volume init" errors.
 */
//#define SPI_SPEED SPI_HALF_SPEED
//#define SPI_SPEED SPI_QUARTER_SPEED

in the config. so I tried SPI_SPEED_5
but while I was compiling I noticed that the firmware reset the SPI_SPEED twice before it finished compiling. Did it do anything to change the #define Spi_speed if the value gets changed twice? Could the spi speed be the problem?

p3p, 2018-04-24:
With SPI speed can always be a problem, especially when used over wires instead of pcb traces and for more than ~10cm, (it wasn't designed for that), smoothieware doesn't officially support external sd cards for this reason.

That bit of code is in the HardwareSPI section and not in use, it's not possible at the moment because of the way the hardware is designed and what Marlin expects to share, that is why we define `LPC_SOFTWARE_SPI` in the pins file, setting SPI_SPEED in the config is correct and will slow down the SoftwareSPI implementation, This is the exact code my Re-ARM uses but configured to use different pins, at full speed and without any problems.

Sorry I'm not being much help on this I wish I could afford to buy the board to help you (and others) out, although MKS doesn't have the best reputation over at smothieware apparently.

Bob-the-Kuhn, 2018-04-26:
I just got in a MKS SBASE v1.3.  I hooked up the REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER per the readme. The LCD's on board SD card worked immediately.

I'll need to think about this some more to see if I can reproduce the problem.

warbunnies, 2018-04-26:
Hi bob, 
Man do i have questions. So the sbase has 4 pins on it that are for the sd card. Did you mess with those at all? are they flipped on or off? Also could you share every file you had to modify for the sd? I would really love to have something to compare to.

Bob-the-Kuhn, 2018-04-26:
I just enabled the SDSUPPORT and the LCD in Configuration.h.

I did use toady's version of bugfix-2.0.x

How about zipping up your Configuration.h, Configuration_adv.h and your pins_MKS_SBASE.h files and attaching them to your reply.  That way I'll know for sure we're on the same page.

You could also post pictures of J8, EXP2, on the MKS and EXP2 on the LCD.

thinkyhead, 2018-04-26:
> What does the varying levels of green mean?

The darkest shade is a first-order define, set directly for the associated LCD controller. The next lightest is set as a result of that, with the next lightest following from _that_… and so on…

warbunnies, 2018-04-26:
[marlin-upload-github.zip](https://github.com/MarlinFirmware/Marlin/files/1953285/marlin-upload-github.zip)

Here are the files. I'll take pictures when I can. So you didn't mess with any of the switches of the sbase?

Bob-the-Kuhn, 2018-04-27:
Your files work on my system if I either comment out **SD_DETECT_PIN** or I set it to the pin I'm using.

It didn't matter if your **SDSS** change was in or out.  The **SS_PIN** is what selects the SD card on the LCD.

warbunnies, 2018-05-27:
[image]
[image]
[image]
[image]

Sorry Its been so long here is how I wired it. Let me know if I did something wrong.

cdedwards, 2018-05-27:
I found that the MKS Sbase v1.3 board I have has a voltage of 4.7v instead of 5v. This caused issues with the REPRAP Graphics board not being bright enough. The trim pot underneath the display had to be adjusted for proper brightness.

warbunnies, 2018-05-28:
Does that have anything to do with be sd card not working...?

warbunnies, 2018-06-03:
I finally found a fix! I'm gonna have to cross test and see if it was one or both of the steps I did but I'll get back with that later today or tomorrow. 

1)moved the pins (miso,sck,mosi) to the 5th stepper motor breakout pins instead of the suggest pins cause it wasnt working. 
2)reformatted the sd card to fat32. 

The main difference between the pin switches was that the lcd would freeze when I put in the sd card. So i figured it was at least trying to read the card finally and stalling. So I reformatted the sd card and it is now printing like a charm. 

Also the sbase really needs more breakout pins... I love 32bit marlin but I might have to find a different 32 bit board.
(seriously the S-curve acceleration is awesome)

thinkyhead, 2018-06-12:
It's possible the suggested SPI pins are in conflict with some other function. What does the `M43` (`PINS_DEBUGGING`) report say about those pins?

warbunnies, 2018-06-12:
Omg i did not know M43 was a thing... darn it that would have been so much easier! >< Its a bit of a busy week for me but ill see if i can find some time to check.

closed 2019-05-16 by thinkyhead