MarlinFirmware/Marlin #10401: [2.0X] upload_disk not defined, compile error on Mac

jokeman3000, 2018-04-13:
NameError: name 'upload_disk' is not defined:
File "/Library/Python/2.7/site-packages/platformio/builder/main.py", line 171:
env.SConscript(item, exports="env")
File "/Users/joseph/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/Users/joseph/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/Users/joseph/Downloads/3D/Firmware/Marlin-bugfix-2.0.x/Marlin/src/HAL/HAL_LPC1768/upload_extra_script.py", line 133:
UPLOAD_PORT = upload_disk

**Successful compile after adding** 

> upload_disk = ""

**to line 9 of upload_extra_script.py**

annego15, 2018-04-13:
for me it worked with that in upload_extra_script.py:

    import os
    drives = os.listdir('/Volumes')  #  human readable names
    target_file_found = False
    target_drive_found = False
    for drive in drives:
      if target_drive in drive and target_drive_found == False:  # set upload if not found target file yet
        target_drive_found = True
        upload_disk = '/Volumes/' + drive + '/'
      if os.path.isdir('/Volumes/' + drive + '/'):
        filenames = os.listdir('/Volumes/' + drive + '/')
        if target_filename in filenames:
          if target_file_found == False:
            upload_disk = '/Volumes/' + drive + '/'
          target_file_found = True
    #
    # set upload_port to drive if found
    #

    if target_file_found == True or target_drive_found == True:
      Import("env")
      env.Replace(
          UPLOAD_PORT = upload_disk
      )`

thinkyhead, 2018-04-14:
I've added the global per your suggestion. Probably Python is being "strict" about this.

closed 2018-04-14 by thinkyhead