MarlinFirmware/Marlin #10402: Second time homing Z axis only goes down.

Ronald-3D, 2018-04-13:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1908425/Configuration.zip)
After successfully print several designs with the 'mesh bed leveling option' configured, I installed an inductive probe and changed the auto bed leveling to Bilineair and configured the probe. After loading the software to the printer (Wanhao D6) the first homing cycle (initiated via the LCD screen) went oke.

1- Bed lowers a little
2- X axis is homed
3- Y axis is homed
4- Hotend move's to te middle of the bed.
5- Bed lowers again a little bit.
6- Then bed moves up until sensor is reached
7- Bed lowers a little (finished)

If I do another home again via the LCD screen the first 5 steps are the same, only instead of moving up in step 6 the bed lowers again and the stops after a short while. The same happens when trying the auto bed leveling. The bed lowers by every measuring point and never goes up, only down.
If I restart the printer, then the first homing sequence goes oke.

I would expect that the bed (z axis) moves up until sensor is reached instead of moving down twice.

Is this a known issue? Tried the latest bugfix 1.1.x, but the same issue.

I included the configuration.h and configuration_adv.h

thinkyhead, 2018-04-14:
It sounds like your probe triggering right away because the bed isn't lowering enough to pass the probe threshold. Play around with these settings…
```cpp
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
//#define Z_AFTER_PROBING           5 // Z position after probing is done

#define Z_PROBE_LOW_POINT          -2 // Farthest distance below the trigger-point to go before stopping

. . .

//#define UNKNOWN_Z_NO_RAISE // Don't raise Z (lower the bed) if Z is "unknown." For beds that fall when Z is powered off.

//#define Z_HOMING_HEIGHT 4  // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
                             // Be sure you have this distance over your Z_MAX_POS in case.
```

Ronald-3D, 2018-04-14:
Problem solved!
The problem were not the settings you suggested. The bed was already far away from the probe during the second homing attend. But your comment about probe triggering triggered me to think. So I changed the pull-up settings for the z-axis end stop and that did the trick.
Homing and probing is working as expected now.

Thanks for your help.
Happy printing.

Ronald-3D, 2018-04-14:
Problem solved!
I could not imagine that it were the settings you suggested because the bed was already far away from the sensor before I started the second homing attend. But your comment about the probe triggering was made me thinking. So I disabled the ENDSTOPPULLUP_ZMIN and ZMIN_PROBE settings in the ENDSTOP SETTINGS section and that did the trick.
Homing and autobed leveling are working now as expected.

Thanks for your help.
Happy printing.

closed 2018-04-14 by Ronald-3D