MarlinFirmware/Marlin #10403: Marlin 1.1.8 overtemperature TMC2130

acocalypso, 2018-04-13:
### Description

Im facing an issue with tmc2130 and Marlin 1.1.8 where the print is canceld after first or second layer due to overhead issue on the z-axis

### Steps to Reproduce

1. Start the print
2. after auto leveling print starts
3. first layer finish z moves up a bit fast and then the print stopps.

**Expected behavior:** print continous till done.

**Actual behavior:** print stops after first/second layer

#### Additional Information

I used Marlin 2.x before and didn't faced that problem.

I used Z with 1100 mA

and tested with 500,600,700,800,900,1000

log M122, configuration.h and configuration_adv.h in attachment.
[log.zip](https://github.com/MarlinFirmware/Marlin/files/1908796/log.zip)

teemuatlut, 2018-04-13:
You can pick from:
1. Increase cooling
2. Lower current. 1.1A_rms is really high when the rated max is 1.2A_rms.
3. Keep using Marlin 2.0. Although there really shouldn't be any difference as far as TMC drivers are concerned.

If you just want to get rid of the error, disable `STOP_ON_ERROR`. The feature was implemented in case a driver shuts down trying to protect itself and Marlin keeps on printing unaware that one of the stepper motors doesn't move anymore.

acocalypso, 2018-04-13:
thanks, I will test it out and get back to if its working or not.

Lyr3x, 2018-04-14:
@teemuatlut should we maybe extend the http://marlinfw.org/docs/features/tmc_drivers.html ?
Most of the issues opened about the tmc2130 are discussed 100 times before. I'm glad if i can help others to get the tmcs running, as u helped me month ago... But we always repeating talking about the same stuff

acocalypso, 2018-04-14:
After reducing the mA its working like a Charme. Sorry for bothering you.

closed 2018-04-14 by acocalypso