MarlinFirmware/Marlin #10410: bugfix-1.1.x: stuttering motion

HWiese1980, 2018-04-14:
Hi everyone,

since, I think, yesterday I've been experiencing stuttering motion on my RAMPS 1.4/Marlin based printer (see https://www.youtube.com/watch?v=CbQdjndpYNM). I'm not sure what I've changed in the configuration that caused this to happen. But I also recognized some weird behavior when homing the axes. There's a short pause of a between hitting the end switch and moving back for the short, slower second homing move, which hadn't been there before. 

I remember having switched on AUTOMATIC_POWER_CONTROL a couple of days ago. I can't remember though if this started to happen immediately after... 

It doesn't matter whether I print from a RasPi using OctoPrint or by USB from my normal PC using Repetier Host. It's the same effect.

Interestingly, before this behavior began, in case of an "abort print" command it took the printer a while to actually abort, like, it had to empty the buffer at first. Now it seems that the abort happens immediately, as if there were no commands in the buffer or no buffer at all. I haven't switched it off though. All buffer settings are still set to upstream repository defaults. 

If you want to take a look at my configuration, I've forked Marlin and push my [Configuration.h](https://github.com/HWiese1980/Marlin/blob/bugfix-1.1.x/Marlin/Configuration.h), [Configuration_adv.h](https://github.com/HWiese1980/Marlin/blob/bugfix-1.1.x/Marlin/Configuration_adv.h) and [pins_RAMPS.h](https://github.com/HWiese1980/Marlin/blob/bugfix-1.1.x/Marlin/pins_RAMPS.h) on my GitHub fork. They are the only files that I've changed. It's based on upstream branch `bugfix-1.1.x`, commit 9769d799df76c1df0cfbeb78c9da02219f8a689c .

Thanks a lot in advance!

HWiese1980, 2018-04-14:
It is definitely related to AUTO_POWER_CONTROL. I've switched it off and the printer works perfectly again.

teemuatlut, 2018-04-14:
Likely related to the TMC2130 not getting initialized properly when using `AUTO_POWER_CONTROL`.

HWiese1980, 2018-04-14:
The PSU pin only switches on power for my heat bed. The stepper drivers (VM) are powered by a second PSU that is permanently on. I only want to use power control because the heat bed PSU makes some fan noise that I find annoying after a while. If that wasn't the case I'd completely leave the PSUs on. 

Might that be the reason? VM permanently on while AUTO_POWER_CONTROL and the TMC2130 expect VM to be off sometimes?

HWiese1980, 2018-04-14:
Oh, and to add to that: Z-axis, which contrary to X and Y has only a TMC2100, shows the same *homing* behavior.

teemuatlut, 2018-04-14:
If the drivers have VM when writing new register values then the communication should be ok.
But IIRC, `AUTO_POWER_CONTROL` should try to push configs to the drivers when it switched PSU on.

If you can, switch out the TMC2130 drivers with something that doesn't require communication, like TMC2100 or A4988 just to rule them out.

HWiese1980, 2018-04-14:
I'll see. I have some spare TMC2100 and A4988 floating around. Currently the printer is working. When it's done, I'll go check whether it's TMC2130 related.

thinkyhead, 2018-04-18:
I've patched `bugfix-1.1.x` to make sure there's no extra delay added if the steppers are already enabled. Please test the latest code to see if it fixes the issue.

HWiese1980, 2018-04-18:
Hmm... I'm getting an error message when trying to compile the latest commit:

    sketch\Marlin_main.cpp: In function 'void set_bed_leveling_enabled(bool)':

    Marlin_main.cpp:2472: error: no matching function for call to 'unified_bed_leveling::get_z_correction(const float&, const float&, const float&)'

                 current_position[Z_AXIS] += ubl.get_z_correction(pos[X_AXIS], pos[Y_AXIS], pos[Z_AXIS]);

                                                                                                       ^

    sketch\Marlin_main.cpp:2472:99: note: candidate is:

    In file included from sketch\Marlin_main.cpp:363:0:

    sketch\ubl.h:284:18: note: static float unified_bed_leveling::get_z_correction(const float&, const float&)

         static float get_z_correction(const float &rx0, const float &ry0) {

                      ^

    sketch\ubl.h:284:18: note:   candidate expects 2 arguments, 3 provided

    Marlin_main.cpp:2477: error: no matching function for call to 'unified_bed_leveling::get_z_correction(const float&, const float&, const float&)'

                 current_position[Z_AXIS] -= ubl.get_z_correction(pos[X_AXIS], pos[Y_AXIS], pos[Z_AXIS]);

                                                                                                       ^

    sketch\Marlin_main.cpp:2477:99: note: candidate is:

    In file included from sketch\Marlin_main.cpp:363:0:

    sketch\ubl.h:284:18: note: static float unified_bed_leveling::get_z_correction(const float&, const float&)

         static float get_z_correction(const float &rx0, const float &ry0) {

                      ^

    sketch\ubl.h:284:18: note:   candidate expects 2 arguments, 3 provided

    exit status 1
    no matching function for call to 'unified_bed_leveling::get_z_correction(const float&, const float&, const float&)'

HWiese1980, 2018-04-19:
I confirm your fix working. Axes move smoothly now with AUTO_POWER_CONTROL enabled. Thanks a lot! Keep rocking!

commit 9631a3ec (2018-04-17): Prevent delay when power is already on

Fix #10410
commit 3c826e5d (2018-04-17): Prevent delay when power is already on

Fix #10410
commit 444b6d6d (2018-04-17): Prevent delay when power is already on

Fix #10410
closed 2018-04-19 by HWiese1980