MarlinFirmware/Marlin #10413: [2.0.x] Compilation error on 2560

ekozan, 2018-04-14:
Hello 
 
Since eab3a89a82258641919a2aa03bc4900bdb2b9e63 we got an error on compilation 

```
In file included from sketch/src/lcd/lcdprint_u8g.cpp:26:0:
sketch/src/lcd/u8g_fontutf8.h:12:23: fatal error: clib/utf8.h: No such file or directory
 #include "clib/utf8.h"
                       ^
```

it's  on travis too https://travis-ci.org/MarlinFirmware/Marlin/builds/366418389#L614

```
Marlin/src/lcd/u8g_fontutf8.c:11:22: fatal error: clib/u8g.h: No such file or directory
#include "clib/u8g.h"
```

comps, 2018-04-14:
Duplicate of #10395 ?

waddyano, 2018-04-14:
Just hit the same thing - LPC1768 build

ekozan, 2018-04-15:
adressed by https://github.com/MarlinFirmware/Marlin/pull/10419

closed 2018-04-15 by ekozan