MarlinFirmware/Marlin #10415: HAL.h missing error

isaacfank, 2018-04-14:
### Description

I am trying to compile on both arduino ide and atom and it gives me the same error. It gives the error if i have bed leveling and eeprom turned on and if i have it turned off. Looking at the error, there really is no file named hal.h in that same folder, but there is a hal.h in the hal_DUE folder, so i dont know what to do from there.

### Steps to Reproduce

1. Hit compile
2. Read error log
3. Cry
[P3Steel-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/1910752/P3Steel-2.0.x.zip)

p3p, 2018-04-14:
We recently changed the way HAL headers were included should be an easy fix.

p3p, 2018-04-14:
#10416 should fix this issue, but you may still have problems building because of an ongoing issue with` u8g.h`

isaacfank, 2018-04-14:
Ok I'll check it out. I don't think I'll have an issue because I don't have a screen on mine. (that's what you're referring to I think.)
Isaac Fankhauser
----------------------
IT/Network Admin
CNC Programmer
3D Design and Printing

---- On Sat, 14 Apr 2018 17:25:52 -0400 notifications@github.com wrote ----

#10416 should fix this issue, but you may still have problems building because of an ongoing issue withu8g.h

closed 2018-04-16 by thinkyhead