MarlinFirmware/Marlin #10418: z-height is incorrect after G28 (using inductive probe)

coldpower, 2018-04-15:
### Description

I have FLSUN i3 (2016 model) with board BT7272A and trying to use marlin 1.1.8 release.

Using inductive probe connected to z-min endstop.
G28 works fine, but nozzle is too high. Height of nozzle matches to Z_MIN_POS from the configuration.h, i.e. i have
#define Z_MIN_POS -3
in my configuration.h and nozzle remains 3mm above headbed.

Previously used marlin 1.1.0-RC8 with no issues.

### Steps to Reproduce

1. run G28
2. run G0 X100 Y100 Z0
3. observe nozzle position - it is still 3mm above the bed

**Expected behavior:** 
nozzle to be close to the bed (0.1mm or so)

**Actual behavior:**
nozzle is 3mm higher than it should be

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1911520/configuration.zip)

thinkyhead, 2018-04-15:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

coldpower, 2018-04-16:
Hi @thinkyhead, 

thank you for the response.
I can reproduce the issue with bugfix-1.1.x as well.

thinkyhead, 2018-04-16:
I assume you want the position after homing to be 0 even though you also want to be able to move the nozzle below the bed for various reasons. If that's the case then you should have enabled this setting in your `Configuration.h`:
```
#define MANUAL_Z_HOME_POS 0
```

thinkyhead, 2018-04-16:
Note that in the latest code we automatically disable software endstops during manual probing procedures and then re-enable them afterwards. So there's no more need with Marlin 1.1.9 / 2.0 to set `Z_MIN_POS` to -3 for that kind of usage.

coldpower, 2018-04-16:
okie, thanks for the advice.

Actually i've found a way to work in my scenario - i just disabled software endstops (M211) and was able to move nozzle below the homing point.

So, there are no more issues from my side, ticket can be closed.

Thanks a lot!

closed 2018-04-18 by thinkyhead