MarlinFirmware/Marlin #10420: Double Zero

thjubeck, 2018-04-15:
When Manually stepping the z axis in .1 increments, the LCD shows that the axis goes to zero then the next step is also zero then negative zero point one. The axis is physically moving in the correct increments (verified by calipers and movement of the axis) but the zero is repeated on the display

The pattern is as follows 
0.1
0
0
-0.1
Whereas it should be
0.1
0
-0.1
-0.2

I have a prusa i3 on a ramps 1.4 board with the newest marlin release installed.

Roxy-3D, 2018-04-15:
How about more data?   Please start at +.5 and go to -.5.   I'm wondering if we are seeing a floating point round off error.      Going from +.5 to -.5 will let us see more about the movement.

thjubeck, 2018-04-15:
It moves 1.1

Roxy-3D, 2018-04-15:
Does it move 1.1 mm according to the LCD Display?   And does M114 report a 1.1 mm movement?
Because if the nozzle is moving 1.1 mm but the LCD Display gives the correct number it means your steps per unit is defined wrong.

thjubeck, 2018-04-15:
it moves 1 mm according to the display and command but 1.1 according to using calipers to check it. When moving at zero the axis physically moves .1 but the display shows zero for both .1 movements

thjubeck, 2018-04-15:
This only happens when moving across zero all other above and below movements and measurements are good

thinkyhead, 2018-04-15:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thjubeck, 2018-04-16:
Haven't uploaded the bugfix yet but here is the M114 output on both "zeros"
1:37:18.322: X:-28.00 Y:-5.00 Z:0.00 E:0.00 Count X:-2800 Y:-500 Z:0
1:37:33.744: X:-28.00 Y:-5.00 Z:-0.10 E:0.00 Count X:-2800 Y:-500 Z:-160

thjubeck, 2018-04-16:
Going to try the bug fix in a min just thought I would share that

thjubeck, 2018-04-16:
Same issue with the bugfix
this is going from point one to the first "Zero" to the next then to negative point one
1:46:10.130: X:-28.00 Y:-5.00 Z:0.20 E:0.00 Count X:-2800 Y:-500 Z:320
1:46:20.544: X:-28.00 Y:-5.00 Z:0.10 E:0.00 Count X:-2800 Y:-500 Z:160
1:46:45.130: X:-28.00 Y:-5.00 Z:-0.00 E:0.00 Count X:-2800 Y:-500 Z:0
1:47:08.607: X:-28.00 Y:-5.00 Z:-0.10 E:0.00 Count X:-2800 Y:-500 Z:-160

thjubeck, 2018-04-16:
All axis actually. Didn't check before.

thinkyhead, 2018-04-16:
And always (and only) on zero, eh? That's a clue, but a clue to what I'm not sure.

thjubeck, 2018-04-16:
Yeah only when crossing zero. Just figured out possibly what it is. if I step it through the PC interface it goes to -.09 after it crosses zero. Above it it is okay but as soon as I hit zero it goes 0.0 then -0.09 and the same if I set it to go to -0.3 then step it through the interface it goes to .09 after crossing zero going positive.

thinkyhead, 2018-04-16:
Hmm, see if the code in this branch works any better:
https://github.com/thinkyhead/Marlin/archive/bf1_fix_and_improve.zip

thjubeck, 2018-04-16:
Still the same.

thinkyhead, 2018-04-16:
Ok, I'll keep looking into it.

thinkyhead, 2018-04-16:
> Yeah only when crossing zero. Just figured out possibly what it is. If I step it through the PC interface it goes to -.09 after it crosses zero. Above it it is okay but as soon as I hit zero it goes [...]

I find this hard to comprehend. Can you restate?

thjubeck, 2018-04-16:
I tried using repetier to step it manually instead of the encoder and LCD menu. I set the axis to .3 exactly then start moving down manually by .1. When I reach zero the next increment goes to -.09 instead of -.1  according to the displayed numbers on the main info screen on the LCD.

thinkyhead, 2018-04-16:
Ah, so it's float rounding. It's actually 0.099999. That's something I know how to fix!

thinkyhead, 2018-04-16:
Try [this](https://github.com/thinkyhead/Marlin/archive/bf1_fix_and_improve.zip) again and see if the rounding works any better.

thjubeck, 2018-04-23:
Can't download it. Sorry for the long reply something went haywire after switching firmwares so many times. Think my EEprom was corrupted possibly anyways new ramps and arduino so I will try it whenever you get it uploaded.

thinkyhead, 2018-04-24:
Changes have been merged, so just grab the latest code.

thjubeck, 2018-04-28:
Finally got bugfix 2.0 to work ( the compile name too long issue) but the double zero is still there..

Bob-the-Kuhn, 2018-04-29:
Try this file.

I was able to reproduce the issue when using a LCD and movig the axis in 0.1mm steps.

The LCD's float to string routine truncates the number at the last displayed digit rather than rounding.  When I added some more decimal places the display showed 
0.200
0.100
0.000
-0.099
-0.199

[utility.zip](https://github.com/MarlinFirmware/Marlin/files/1958318/utility.zip)

thjubeck, 2018-04-29:
Yep mine does the same so I guess the better question is the case is why does it do this only after crossing zero?

thinkyhead, 2018-04-29:
Because of the way floating-point works.

thinkyhead, 2018-04-29:
Try this change to `ultralcd.cpp`:
```diff
    if (lcdDrawUpdate) {
      const float pos = NATIVE_TO_LOGICAL(processing_manual_move ? destination[axis] : current_position[axis]
        #if IS_KINEMATIC
          + manual_move_offset
        #endif
      , axis);
-     lcd_implementation_drawedit(name, move_menu_scale >= 0.1 ? ftostr41sign(pos) : ftostr43sign(pos));
+     lcd_implementation_drawedit(name, move_menu_scale >= 0.1 ? ftostr41sign(FIXFLOAT(pos)) : ftostr43sign(FIXFLOAT(pos)));
    }
```

Bob-the-Kuhn, 2018-04-29:
Did the new utility fix the problem?

thinkyhead, 2018-04-29:
To deal with it universally it's simplest to apply the updated `FIXFLOAT` macro to all the `utility.*` string conversion functions. The extra logic there won't have any major impact since these functions are used infrequently.

thjubeck, 2018-04-29:
Sorry to be a bother but that didn't work downloaded the newest bugfix https://photos.google.com/share/AF1QipPfYc_uckSNWhqct3L-dGgoZFKGt1Zv_XIyi_VR2y7FtWPirNdAQ9gzZ9erngsFhQ?key=bXZrVDlOMGU4ZkRkRGRTRFNMM1ZyM3VPUkxDdTZn

thinkyhead, 2018-04-29:
Then… we'll just keep on tweaking it till it works.

Bob-the-Kuhn, 2018-04-29:
Just in case ...

Here's the low level routines with rounding.

[utility (2).zip](https://github.com/MarlinFirmware/Marlin/files/1958943/utility.2.zip)

thinkyhead, 2018-04-30:
Updated the patch, so hopefully this covers it!

boelle, 2019-02-20:
@thjubeck was the problem solved in latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

boelle, 2019-03-17:
i cant reproduce this one

closed 2019-03-17 by p3p