MarlinFirmware/Marlin #10421: [2.0.x] Rough stepping in curves on AZSMZ board

alexyu132, 2018-04-15:
### Description

Running Marlin 2.0 on my AZSMZ board driving a CoreXY printer, the steppers run very roughly in certain curves (making loud grinding noises). The board is running LV8729 drivers set at 1/16 microstepping. The problem doesn't affect longer fast moves, but is more prevalent in curves with many segments. At 4000 acceleration and 150mm/s print speeds, the problem is enough to cause layer shifts, but the issue is definitely present at lower speeds and accelerations as well. 

### Steps to Reproduce

Running the attached gcode snippet produces the issue (consists mainly of gentle curves mostly at  70-140mm/s, with some 300mm/s travels). All testing was done printing from SD connected to an attached AZSMZ 12864 LCD controller.
[F_Gimball_Test_Print.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/1911715/F_Gimball_Test_Print.gcode.txt)

Problem tested present with accelerations of 500, 1000, 3000, 4000, 9000 and jerks of 1, 5, 10, 20. The problem is less present with lower accelerations, though lower jerk doesn't appear to reduce the grinding noises. Changing min stepper pulse time from 0-2 has no noticeable effect. I also tried forcing quadstep mode but the grinding was still present.

#### Additional Information
[Video of the issue](https://www.youtube.com/watch?v=SLRal0E1-Iw)
[configs](https://github.com/MarlinFirmware/Marlin/files/1911719/configs.zip)

boelle, 2019-02-20:
@alexyu132 is this still an issue with latest bugfix 2.0?

are you sure its not a hardware issue?

alexyu132, 2019-02-20:
This problem was caused by an issue with the planner that seemed to be causing stuttering and pauses during short segment moves. The planner optimization commits last year seemed to have fixed the problem and I haven't had any rough movement since.

boelle, 2019-02-20:
@alexyu132 then maybe click the green close button below

closed 2019-02-20 by alexyu132