MarlinFirmware/Marlin #10422: FR - Adjust Diagonal rod separately - Delta Printer

Zezoh, 2018-04-15:
Thank you for your great work, 

Is there a possibility to add  a way to change the diagonal rod length for each tower using LCD instead of doing it in Configuration.h file ( //#define DELTA_DIAGONAL_ROD_TRIM_TOWER { 0.0, 0.0, 0.0 })

Steps to Follow: 

1. Using G33 for Delta Auto-Calibration 
2. Print a testing model it could be a hexagon shape
3. Measure hexa model and adjust each rod separately with LCD. 

Regards,

thinkyhead, 2018-04-16:
Enable `DELTA_CALIBRATION_MENU` and you'll find it at the bottom of the "Delta Calibration" menu.

Zezoh, 2018-04-16:
Thanks Scott,

LVD-AC, 2018-04-16:
> Print a testing model it could be a hexagon shape
> Measure hexa model and adjust each rod separately with LCD.

Dimensional errors are most likely NOT caused by a wrong diagonal rod value. You better know what you're doing before messing with it. Just for this reason I'm not in favor to make DELTA_DIAGONAL_ROD_TRIM_TOWER changeable to easy (G-code, LCD).

Zezoh, 2018-04-16:
Dear Luc, thank you for your feedback, can you please provide more details about the causes of   dimensional errors,

LVD-AC, 2018-04-16:
Diagonal rod length is (should be) pretty accurately know; applying the rather large corrections required to compensate for dimensional errors is just making sure you get the rod length wrong.

cjsoong, 2018-04-18:
@Zezoh You can try my modified version to meet your needs.
DELTA_DIAGONAL_ROD_TRIM_TOWER & DELTA_RADIUS_TRIM_TOWER  adjustable .
[Marlin-bugfix-2.0.x_20180417b.zip](https://github.com/MarlinFirmware/Marlin/files/1925981/Marlin-bugfix-2.0.x_20180417b.zip)

Zezoh, 2018-04-20:
Thanks @cjsoong, I will give it a try

boelle, 2019-02-20:
@Zezoh what was the result?

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

closed 2019-07-11 by thinkyhead

evandene, 2019-12-06:
> @Zezoh You can try my modified version to meet your needs.
> DELTA_DIAGONAL_ROD_TRIM_TOWER & DELTA_RADIUS_TRIM_TOWER adjustable .
> [Marlin-bugfix-2.0.x_20180417b.zip](https://github.com/MarlinFirmware/Marlin/files/1925981/Marlin-bugfix-2.0.x_20180417b.zip)

I was looking up your configuration file and I was wondering how you did set your EXTRUDER feedrate. I couldn't find any steps per unit set for the extruder(s)
Could you help please?