MarlinFirmware/Marlin #10423: Marlin\src\lcd\u8g_fontutf8.c:11:22: fatal error: clib/u8g.h: No such file or directory

lolisakio, 2018-04-15:
Marlin\src\lcd\u8g_fontutf8.c:11:22: fatal error: clib/u8g.h: No such file or directory
this error on bugfix-2.0.x , so 
it should changed 
#include <clib/u8g.h> to #include <lib/u8g.h>

in Marlin\src\lcd\u8g_fontutf8.c and Marlin\src\lcd\u8g_fontutf8.h

Bergerac56, 2018-04-15:
As u8g.h is invoqued by multiple files, it seems easier to add a marlin\clib\ directory and drop u8g.h file in it

waddyano, 2018-04-15:
This PR doesn't seem to fix my local LPC1768 platformio build. But what I did do is similar to the suggestion . I added a clib/u8g.h that just includes lib/u8g.h in a directory already on the include path

Elenedeath, 2018-04-15:
go to file `Marlin\src\lcd\u8g_fontutf8.c` and change` #include <clib/u8g.h>` to `#include <lib/u8g.h>`
work fine for me to compile marlin 2 for lpc1768

thinkyhead, 2018-04-15:
@yhfudev — Is this fixed with our latest patch as of today?

p3p, 2018-04-15:
As long as PlatformIO pulls down a new copy of the library this is fixed, so clean then build.

closed 2018-04-15 by thinkyhead

waddyano, 2018-04-16:
I now have a clean build without addition hacks - thanks everyone - but I had to somewhat beat platformio into submission to download the new version of the library by uninstalling it. It didn't seem to do it automatically even though I could manually download the zip and see it had changed. I tried platformio lib update and also just letting vscode do the build. Is there something else I should have done? I am new to platformio.

thinkyhead, 2018-04-16:
You did good. PlatformIO doesn't download updated libraries automatically.