MarlinFirmware/Marlin #10424: [Bugfix 2.0] Bilinear leveling for Delta

Bergerac56, 2018-04-15:
I started to test marlin on a delta printer (mega2560 AVR, FLSUN type). Printer is correctly configured and works quite fine without any "bed leveling" option. The axes are moving the right way and I can print without "mirror" effect.

When I try bilinear option, it does not work. Marlin issues a (L,R,F,B) out of bonds, whathever the values I can put in configuration.h (But I was able to let G29 work by using G29 L-x Rx Fx B-x)

I have solved this by modifying 2 files in marlin:
In configuration.h
  #define LEFT_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS - MIN_PROBE_EDGE)
  #define RIGHT_PROBE_BED_POSITION (DELTA_PRINTABLE_RADIUS - MIN_PROBE_EDGE)
  **#define FRONT_PROBE_BED_POSITION (DELTA_PRINTABLE_RADIUS - MIN_PROBE_EDGE)
  #define BACK_PROBE_BED_POSITION -(DELTA_PRINTABLE_RADIUS - MIN_PROBE_EDGE)**
=> Signs of "Back" and "Front" have to be changed

And, as sanitycheck will cry if not adapted:
in src/HAL/HAL_AVR/Inc/SanityCheck.h
**static_assert(FRONT_PROBE_BED_**POSITION > BACK**_PROBE_BED_POSITION, "FRONT_PROBE_BED_POSITION must be more  than BACK_PROBE_BED_POSITION.");** 
=> Change the < sign by a > sign

As this is my first steps with a Delta, I do not know if it is a bug or specific to my configuration.

thinkyhead, 2018-04-20:
1. The probe offsets have to be considered in deciding whether a point can be reached.
2. You can leave those values undefined, and they will be set correctly by `Conditionals_post.h`.
3. Front is always negative and back is always positive.

Bergerac56, 2018-04-22:
@thinkyhead Thanks for that. As it works for me with the mods, I did not investigate more why front and back are reversed in my case. The printer is in another location. I will dig in later

closed 2018-04-22 by Bergerac56