MarlinFirmware/Marlin #10425: [1.1.8

mtmello2003, 2018-04-15:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-04-15:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

mtmello2003, 2018-04-22:
please close this trhead, I will open another one. 

Just to make the things organized! :)

Linear advanced is okay right now

closed 2018-04-23 by thinkyhead