MarlinFirmware/Marlin #10435: Mesh Edit UBL Function does not work on delta printers

pfjason, 2018-04-16:
### Description

When clicking "Edit Mesh" in the UBL menu, the printer homes, then tries to send the X axis to 0, then Y to 0,  ignoring the fact that the printer is a delta and nearly breaking my rods. 

### Steps to Reproduce

On 

**Expected behavior:** Functionality to edit the built mesh

**Actual behavior:** X goes to 0 position, sending the effector way outside of the towers, then Y starts to do the same and I emergency powerdown the printer.

Roxy-3D, 2018-04-17:
@pfjason    If the current bugfix doesn't resolve the problem, you should be able to use the M421 command to edit mesh points.   It isn't has nice as using the G29 P4 interactive mesh editing...  But it will allow you to adjust the mesh points to where you need them to be.

@thinkyhead     Perhaps we should remove the UBL G29 P4 capability and make it stand alone for any mesh based leveling system???   Kind of like the G26 was originally for UBL, but now it works for all mesh based systems.

pfjason, 2018-04-17:
My printers will be tied up for a little bit, but I'll test it when I can.

thinkyhead, 2018-04-18:
> Perhaps we should remove the UBL `G29 P4` capability and make it stand alone (and optional) for any mesh based leveling system?

- MBL wouldn't need it.
- For 3-point and planar leveling it would be overkill.
- Bilinear leveling simply uses the reachable area, interpolates for points that can't be reached on a delta, and for square beds assumes that the area outside the probed region will either flatten out (if the area is very large) or that it continues the same slope (when the area is relatively small). This seems to work for the majority of use cases, and it keeps ABL bilinear relatively small and simple. If the extrapolated points (outside the delta circle) are way off, `M421` can be used to adjust them, but this usually isn't needed.

I appreciate the way UBL works, for those who like to make the mesh span the bed, even where the probe can't reach. But I think it's also appreciated to have a leveling system that does everything automatically and doesn't require a mix of auto and manual leveling. And, it would be a pretty big re-work of ABL to combine both auto and manual leveling, essentially it would be a total rewrite of its `G29`.

Roxy-3D, 2018-04-23:
It might help to back up 4 or 5 months and try an older version of Marlin and see if the problem exists.   I think it probably will not exists.    If that is true, we can look at the differences between the older code and the current code with respect to G29 P4.   

The big problem is I don't have a Delta printer so it is difficult to trouble shoot problems without assistance.   I suspect the problem is something small and won't be hard to fix.   But without a little more information, it is going to be difficult to figure this out.

yoch2015, 2018-04-24:
Thank you.
Perhaps the change in # 9510 seems to be affecting, but now we are planning to change it on M421.

thinkyhead, 2018-04-24:
Reverted one part of #9510 labeled as "SHOULD NOT OCCUR" but maybe it COULD OCCUR after all…?

yoch2015, 2018-04-25:
I tested 21f5ca6.
There was no problem.
However, if the Z value drops to a certain value while editing P4, the motion to restore Z back and the motion to lower Z are executed at high speed.
It is like a vibration.

thinkyhead, 2018-04-26:
> if the Z value drops to a certain value

- What is the value of your `Z_CLEARANCE_BETWEEN_PROBES`?
- Are you using the `H` parameter to `G29 P4`?

yoch2015, 2018-04-26:
When running G29P4, the head is about 1 mm away from the glass surface, so lowering the head with the LCD dial will lower the head.
Therefore, the value of Z is close to 0.
H option not attached.

Roxy-3D, 2018-04-29:
And this is just opinion...   But I wouldn't have 1mm of clearance while running the nozzle around an uncalibrated bed.   That just sounds dangerous.    A few seconds to raise and lower the nozzle might make some sense.

boelle, 2019-02-20:
@pfjason did you get a chance to check?

pfjason, 2019-02-20:
Way back in april, on the latest bugfix from way back then, it was not fixed. Since then I've replaced the boards in all my printers and am no longer running Marlin, so I have no idea now.

boelle, 2019-02-20:
maybe consider hitting the green close botton then?

pfjason, 2019-02-20:
Why would I do that? Just because I solved the problem by changing to a non-marlin firmware doesn't mean the bug doesn't exist in Marlin anymore. That's not how bugfixes work.

boelle, 2019-02-20:
i just think its the right thing to do since nobody else have chimed in over a year

but by all means leave it open

closed 2019-08-30 by shitcreek