MarlinFirmware/Marlin #10446: [Bug report] Layer Shifts on bugfix-1.1.x and bugfix-2.0.x.zip

viperchannel, 2018-04-18:
### Description

Random layershifts on different heights on both x and y axis, sometimes on both, sometimes just on one axis, reducing speed, acc and jerk, raising vref does not show any improvement. (tried reducing it to half)
Sometimes it happens on the second layer, sometimes after 2h. I checked all the machanical parts, they are fine and if I go back to a  bugfix version from january it works fine without any shifts.

### Steps to Reproduce

The only differences between my working jan bugfix (as far as I can remember) and the current 1.1.x bugfix and marlin 2.0 bugfix that I made are:

Activating Babystepping
Activating Fade Z
Both config files are attatched, the working Jan version and the 2.0 Version
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1922756/Configs.zip)

#### Additional Information

This is my first Bug report, this is not my native language. I tired my best, let me know what additional information you need.

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-04-18:
Layer shifts are usually a mechanical issue. As far as we know there's no randomness in Marlin. If an issue occurs at _**some point**_ in a G-code file, it will _**always**_ occur at the same point in the file. Can you provide a G-code file that demonstrates this issue consistently?

viperchannel, 2018-04-18:
I don`t think its a gcode issue. The objects in the slicer don`t show that behavior in the preview. Also the point of the shift is always different. If I print the same file multiple times, sometimes it happens in the second layer, sometimes it works fine untill 2h midprint. I printed an object at least 5 times, everytime it was different, sometimes on X, sometimes on Y. Thats wh I don`t think it is a mechanical issue, since the axis change. Then I flashed from 2.0 to the latest 1.1.x bugfix with the same issue. In both cases I reduced Acc from 1600 to 800. raised an lowered Vref, decreased feedrate to 70%. Then I switched back to my january build, printed the same object (had to slice it again, the only setting I changed was the K value in the startcode to a v1.0 value) and it printed absolutely fine.

viperchannel, 2018-04-18:
Here is one of the gcodes I have used. The object had a shift in X direction at around 2.1mm heigth, than another one in the opposite direction 1.63mm later, and after another approx 3.3mm it had a shift in Y direction. 
Before that I printed 2.5kg of the same filament, same kind of objetcs, same slicer settings (exept K value because of V1.0 and 1.5) on my january build with no problems at all. 
[failed object.zip](https://github.com/MarlinFirmware/Marlin/files/1923548/failed.object.zip)

teemuatlut, 2018-04-18:
Could be your drivers overheating. You say you increased cooling for steppers but not to drivers. You also increased Vref that increases heat output.

viperchannel, 2018-04-18:
Sorry, I ment the drivers, not the steppers. I have increased cooling on them, but that was not the reason, eaven with the smaller fan, they did not get noticably warm. I just didn`t know what else to do and wanted to eliminate as many reasons that could cause the problem as possible. Adjusting Vref was just a try, it did not help at all. Motors never got above 45°C btw.

isaacfank, 2018-04-18:
If you have Z fade with a small number and you have a very uneven bed your nozzle will rub on layers and then cause shifting. you could try increasing the M420 to make it fade over a longer distance.

viperchannel, 2018-04-18:
I have also aready though of that. But some shifts happened before reching the fade height and some 20mm after. With a 0.2 layer heigth, thats a lot of layers. I have also tried to move the axis while printing by hand. Thats nearly impossible it has so much torque, i would rip my extruder off the mount. My thought are, that all the calulations like  fade height, bed leveling, linear advanced somhow max out memery or buffer during a print. I have no idea if that is realistic at all.

autonumous, 2018-04-19:
I’m currently investigating a similar issue, trying to determine if firmware related or something else.  
Had two 5 hour+ prints which have suffered layer shifts in the Y direction.  
Initial print was with ABS and had some warping, so i thought might have been with the nozzle hitting the print.  2nd attempt, resliced and used PLA, same result.  layer shifts at different heights, always in the Y direction

I'm trying to replicate with simpler, quicker prints that i can observe.    All ok so far.

Printer has been fine previously.

thinkyhead, 2018-04-19:
Keep experimenting. Disable various features. Test various settings for acceleration and jerk. Etc. The most important thing will be to narrow it down to a single causative factor.

viperchannel, 2018-04-19:
I can already tell, that is is 100% caused by the firmware or one of its functions.
I did not have a single print without shifts on the newer firmwares and as soon as I switched back to the old one, I didn`t have single shift. 
Had it printing for 16h yesterday on various prints, including identical gcodes that had shifts with the newer versions with the exact same mechanical, acc, jerk, vref etc settings and not getting a single shift while using the old version.

Jartza, 2018-04-19:
I can confirm this with the 2.0.x bugfix -branch. After updating from 1.1.5 there is sometimes "loud thunk" in the middle of print (my printer if very silent normally and these "thunks" didn't happen with 1.1.5) and I'm also experiencing **layer shifts**, even with prints speeds of 50mm/s (1.1.5 goes fine with 80mm/s). I'm sure there is no mechanical failure or problem with stepper controller reference voltages, I've played around with different accel & jerk speeds. I even tried the BEZIER jerk and while it makes the overall printing smoother, the erratic moves are there.

I try to experiment more during weekend and if possible, make some comparisons with different values.

LichtiMC, 2018-04-19:
Finally an issue regarding the layer shifts!
I'm experiencing the very same behavior...

teemuatlut, 2018-04-19:
It's _possible_ that I'm seeing the same behavior.
I'll need to start eliminating commits and see which change introduced this. If any.

If you guys want to try something, turn off `MONITOR_DRIVER_STATUS`. That way there will be no communication to the driver after the initial setup.

viperchannel, 2018-04-19:
I am kind of happy, I am not the only one that has those issues. I am currently running the TMC2208 on my printer btw.

LichtiMC, 2018-04-19:
OK, may be, just saying I never had any of these problems with the A4988 drivers.
This could be because of the less torque of stealthchop though...

Jartza, 2018-04-19:
> Also judging by the initial report, this is not isolated to TMC drivers.

Correct, my other printer (Tevo Black widow) has drv8825 and the same issue seems to bother there too, erratic looking moves that are louder than normal printing, although no layer shifts (yet) as I'm driving steppers with more force.

Jartza, 2018-04-19:
[image]

Same gcode-file with 1.1.5 (except modified lin_advance value):

[image]

ejtagle, 2018-04-19:
@Jartza: Could you post your 
DEFAULT_AXIS_STEPS_PER_UNIT and your
DEFAULT_MAX_FEEDRATE ?

A long time ago, when i switched to the bugfix2.0.0 branch, for my printer, i need for the Z axis
STEPS_PER_UNIT = 4000, and if i set MAX_FEEDRATE to 3 or more, the Z axis does not move, but it does a horrible noise (brrrr, but with some discontinuities). I am pretty sure there is some overflow in computations somewhere, never investigated them. Maybe that is being triggered somehow in some strange case also

The way to reproduce always reproduce it is:
#define DEFAULT_AXIS_STEPS_PER_UNIT     { 80, 80, 4000, 97 }
#define DEFAULT_MAX_FEEDRATE            { 300, 300, 4, 30 }

And try to move the Z axis with the LCD menu...

Jartza, 2018-04-19:
`#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 96 }
#define DEFAULT_MAX_FEEDRATE          { 150, 150, 8, 50 }`

There is limit how much 8-bit cards can output steps. 

I also have `#define MINIMUM_STEPPER_PULSE 2`

Jartza, 2018-04-19:
Printing regular 20x20x20 calibration cube, there's almost always now loud "thunk" also when the the perimeter has been printed (there is just very short move command). I tried to record it on video, but the "thunk" is so low frequency that cellphone mic didn't catch it but it's very audible by ear :)

isaacfank, 2018-04-19:
I have my current printer on an old build of 2.0 and its running fine, so ill upgrade it to the latest and let you know if there is any clunking/shifting or whatnot. Ill look, but i might be able to figure out the date i downloaded it and ill know what commit its on.

Jartza, 2018-04-19:
My 2.0.x is less than week old, but I will update later today to latest bugfix-2.0.x to see if it helps any.

Jartza, 2018-04-19:
> Ill look, but i might be able to figure out the date i downloaded it and ill know what commit its on.

Try `git log -n1` to see the last commit :)

viperchannel, 2018-04-19:
My 2.0 was april 11th, the 1.1.x bugfix was april 17th.

viperchannel, 2018-04-19:
By looking at Jartzas pic, i noticed that my shift is different. It looks like his print shifter over several layers. It looks like the little house started to continously move to the left. My shifts are just one layer. For example, it starts the print, after several perfect layers, it shifts in one direction and stays there for a while, untill the next shifts happens. I dont have shifts that stack up on several layers like on the pic

Jartza, 2018-04-19:
@viperchannel that's the most rad example. I searched for mechanical failure and tested belts, wheels, bearings etc. from my printer because I had sudden layer shifts in 1-2 layers in some prints and first thought was of course mechanical/electrical issue. After checking everything, I reflashed 1.1.5 and redid the prints with same gcode without any issues (or clunking)

gloomyandy, 2018-04-19:
Are you all using UBL? If so you might want to try the very latest (as of a couple of hours ago) bugfix version. There has just been a fix made to the way that UBL handles moves that may have some impact on this (but may not).

viperchannel, 2018-04-19:
I was using #define AUTO_BED_LEVELING_BILINEAR on all versions.

ejtagle, 2018-04-20:
@Jartza I would agree with you, but i am using an Arduino Due (32bits) and the problem is exactly the same. If i get some free time, i´ll start debugging...

thinkyhead, 2018-04-20:
> I can already tell, that is is 100% caused by the firmware or one of its functions.
> My 2.0 was april 11th, the 1.1.x bugfix was april 17th.

I'm unable to reproduce the issue, unfortunately. I'll see if adding enabled features from your configs causes the issue to appear and if I'm able to get the issue to manifest then I'll be able to start narrowing it down to some point in time, and maybe a specific commit.

> I reflashed 1.1.5

Many changes since 1.1.5, but I'll do a comparison and see if anything stands out. If you can determine the most recent version or commit where the problem is not present, that will be super helpful.

thinkyhead, 2018-04-20:
Have you tested with `LIN_ADVANCE` disabled to see if that feature is contributing?

viperchannel, 2018-04-20:
Not yet tested without Lin adv. I will give it a try over the weekend if I find some time.

Jartza, 2018-04-20:
@thinkyhead by disabling `LIN_ADVANCE` do you mean completely removing the define, or just setting the value to 0?

thinkyhead, 2018-04-20:
Try both, as time allows.

Grimshadows, 2018-04-21:
This is definitely a firmware issue, the machine is randomly going full speed ignoring jerk in tight spaces. It just goes to full speed in the Y direction only. I can tell that it happened from the loud "thunk" that penetrates my walls and vibrates the floor. This started happening when testing non linear prints. It wont do it on prints that simply move in straight lines. If the print has a curve, like a pipe or tunnel whos diameter is on the horizontal plane it will do it. I can visually see that the machine it self is moving faster in certain palces than I set it too. 

[image]

I can send you the file i am trying to print or gCode and my configuration files if you need. There is something about this firmware that it's not obeying speed and jerk limits and it's torque ripping the Y axis out of step.

let me know what you need, I think I can say this is 100% repeatable at this point for me. Even switched out stepper drivers, got tons of them laying around since it's cheaper to buy them in lots of 4. Standard A4988.

I am really going to hate downgrading firmware for now, all these new features are nifty to have.

autonumous, 2018-04-21:
that explains why I’m not able to reproduce with simple models.  Simple cubes came out fine.  Just tried a benchy had had a very slight layer shift, whereas the “curvy” model i was previously printing had many many layer shift.

Grimshadows, 2018-04-21:
Yep simple cubes come out 100% fine, add curves in tight spaces with walls it goes bonkers.
All this prints just fine on 1.1.8:
[image]

thinkyhead, 2018-04-21:
> This is definitely a firmware issue

Sure, but what _kind_ of issue, and where does it originate? If it were a logical bug in the firmware's motion code, you would not see random behavior, but you should expect to see layer shifts at exactly the same points in the G-code. And there would be more widespread reports of the same issue, which so far we don't have.

From what's been described, I can only say that _something mysterious_ about the bugfix branch is exacerbating an issue with the stepper drivers, causing them to overheat and go into thermal shutdown (which gives instant stops _-thunk!-_ followed by instant restarts after a fraction of a second). See if pointing a fan directly onto the stepper drivers reduces the frequency of layer shifts. If that helps, it tells us something. And if it doesn't, that also tells us something.

thinkyhead, 2018-04-21:
> I can send you the file i am trying to print or gCode

@Grimshadows — Please do! I would like to test with the G-code that gives you the most trouble — that cylinder, for example.

viperchannel, 2018-04-21:
Some reported they are using the 2130 tmc. Wouldn't they stop the print when overheating and report a driver error?
I have also already done the cooling experiment and switched from a 40mm fan to a 120mm fan and still had the shifts.

thinkyhead, 2018-04-21:
(comment deleted after reviewing the thread)

Grimshadows, 2018-04-21:
@thinkyhead 
[Skimmer Intake Venturi.txt](https://github.com/MarlinFirmware/Marlin/files/1934378/Skimmer.Intake.Venturi.txt)

thinkyhead, 2018-04-21:
Thanks! I will test it later today and see if I can reproduce the issue. I hope that I can, really, because then we'll all be on the same page!

Jartza, 2018-04-21:
@Grimshadows I "dry printed" the gcode (without filament and temps to 0) and I think that's very good model to find out the problem as the printer really shakes violently with that gcode. Maybe I should test print with filament too 😆

Jartza, 2018-04-21:
@thinkyhead looks like disabling LIN_ADVANCE either by setting it to 0.00 or undefining LIN_ADVANCE has no effect on this...

Grimshadows, 2018-04-21:
@Jartza Those speed settings are really really toned down from the speeds this thing normally runs at. I toned them down thinking it was a speed issue. Doing a run with simplify now at my normal speeds, lets see if it's the slicing engine. i'll give an update in about 35 min. violently shaking is ok as long as im not hitting the chassis harmonic frequency.

viperchannel, 2018-04-21:
I am pretty sure, my motors were much warmer on the bugfix versions than they normally are. If you are switching between versions, take a look at that. I cannon play around this weekend, I have some prints I need to get finished first.

Grimshadows, 2018-04-21:
[Skimmer intake-S3D.txt](https://github.com/MarlinFirmware/Marlin/files/1934444/Skimmer.intake-S3D.txt)

Ok, ran the print in Simplify.... So... it may not be the Firmware after all not one skip. I don't even have S3D setup for fine tuning and calibrated for this level of detail. i mainly have it set up for speed. So there must be something that Cura is doing different with the slicer than S3D. I think I am going to calibrate S3D at this point and run with it. I don't want to give up the features gained in 1.1.8. Still doesn't explain why linear shapes print fine on Cura but not curved objects.

Video Link if your curious,:
https://youtu.be/FyQbR1NfjtM
yes... my tripod is a 3.5 inch VISE.

viperchannel, 2018-04-21:
I did all my files in simply and had the skips. So it's not Cura that causes it

autonumous, 2018-04-21:
All my simple test prints seem to be fine.  The kids have more benchys, and cubes  than they know what to do with! :-)

Just tried to print a whistle from thingiverse, and suffered several shifts.  

As always, slicer was S3D.  only other big difference is that i’ve just printer at 0.2 layer height, my tests have been 0.3.   So will continue to use this same profile/process for other tests and try to reliably reproduce the issue.

autonumous, 2018-04-21:
My first print noticing layer shifts.  

(These prints were fished from the bin, the support had already been removed

[image]
[image]

I changed the material, thinking the warping of ABS may have knocked the print head off.  Also changed the orientation of the part.  Still same result.

[image]

This was today.  3 distinct layer shifts.  all in the Y direction.
[image]

I’ve swapped  X and Y drivers (A4988) over and restored the same print to see if i get shifts in the X direction or continues with Y, or if it comes out ok.  

Will report back once complete.  depending on results then, will try different firmware options/versions

autonumous, 2018-04-21:
Print completed, initially was looking good, but got a very slight layer shift at a higher point than the previous.  Still in the Y direction, despite swapping drivers round.  Everything else the same.  Same code file, (sliced in S3D).

I wasn’t around at the time it occured

Next will try with a previous firmware version, 

[image]
[image]

autonumous, 2018-04-21:
Print completed with no layer shift at all when using a previously compiled 1.1.8 

[image]

A process of elimination  now.....

Grimshadows, 2018-04-21:
@autonumous I can see from the pictures you are getting the same over extrision on skips that i am seeing on my prints.

Grimshadows, 2018-04-21:
@autonumous Is your gCode using Flips between G0 and G1 commands like mine? Could you post it?

thinkyhead, 2018-04-22:
I've reverted certain portions of the planner back to the way they were in 1.1.8 in this branch:

https://github.com/thinkyhead/Marlin/archive/bf1_revert_planner_test.zip

Test with one of the troublesome objects and see if it causes failure. Stop the print as soon as you get any layer shifting to save on material. You can also use `M111 S8` to set "dry run" mode, where it won't print, but will just do all the other moves. If you don't see the issue in "dry run" mode but you do in normal printing, that would be another clue.

Grimshadows, 2018-04-22:
Dry Run wont work as the skips are sometimes so minute they are not audible. I did a quick print without any real re-calibration. Seems to have fixed the big skips, there were two minor shifts (could be user error, so I will have to retest). I am going to re-calibrate the machine when I get back home from work and try again with this build. Seems to have fixed the glaring skips. The Filament no longer attempts to curl up in the extruder too. I will put the part under the magnifier later and count the layers to see what line of code that relates to. I am still thinking it has to do with those G0 to G1 commands Cura slicer is putting out.

thinkyhead, 2018-04-22:
> Dry Run wont work as the skips are sometimes so minute they are not audible.

I see. So you need the object to see that the axis has gotten shifted.

> I am still thinking it has to do with those G0 to G1 commands Cura slicer is putting out.

On Cartesians, `G0` and `G1` are synonymous in Marlin and will produce exactly the same movement. But there might be differences in what the slicer outputs for each one. For example, do the `G0` commands in the G-code have a different `Fxxx` feedrate than the `G1` commands?

LichtiMC, 2018-04-22:
I maybe found out how to provoke this "loud thunk" and, associated therewith, the skipped steps:

In OctoPrint Control Tab, if using following settings and pressing the arrow keys fast in a row, the motors are loosing steps:
[image]

If I fast press 2 times Y+ then go back to 0 and repeat this process for 4-5 times, my hotend will crash into Y0.
This won't ever happen, if I press Y+ slower. It also wont happen, if I alternately press the other directions, no matter how fast. Only if the same direction is pressed multiple times in a row the problem occurs.

The feedrate is set to 6000 in OctoPrint.

Could be completely unrelated, but maybe it helps.

felenna, 2018-04-22:
Did any one tested it with stealthchop off ?? I heared several stories it then stops layershifting. I got also layershifts.I cranked up the current, lowered the speed  and stuff like that but it wont help. Also the little 20 by 20 cube after adjusting some things there good now.I got a 80 mm fan  cooling the steppers and i got the chineese steppers but the same results as you guys using marlin bugfix 1.1.8 im guessing about 1and a half month old version.  And i slice it with S3D. I also got some weird  movements between printing

[313 more replies]

closed 2018-08-15 by thinkyhead