MarlinFirmware/Marlin #10447: The file or extension is too long. Error

isaacfank, 2018-04-18:
### Description

Compiling for the due with smart ramps.

fork/exec C:\Users\*****\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc.exe: The filename or extension is too long.

### Steps to Reproduce

1. Hit compile
2. Wait
3. Sigh

**Expected behavior:** [What you expect to happen]
I expected it to either have a glib error or compile correctly.
**Actual behavior:** [What actually happens]

fork/exec C:\Users\*****\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc.exe: The filename or extension is too long.

#### Additional Information

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/1923815/Configurations.zip)

p3p, 2018-04-18:
You need to use Arduino 1.9 beta to build Marlin 2.0,

isaacfank, 2018-04-18:
Oh really? It should probably say that somewhere in the readme.

closed 2018-04-18 by isaacfank