MarlinFirmware/Marlin #10451: [1.1.x] #define CUSTOM_USER_MENUS can this function be used to alter xyz offsets?

bruce356, 2018-04-18:
Mendel style Cartesian Dual-X Carriage printer, RUMBA board and RepRap Discount Full Graphic Smart Controller. CUSTOM_USER_MENUS is a brilliant idea (by Tannoo and others).

Hi, looking for help (not sure if anyone is interested, I am not a programmer) to create a GLCD menu item to view current offsets and to be able to change them for XYZ (M218 command) and then store updated to eeprom.

Currently I set these initially in configuration.h
#define HOTEND_OFFSET_X {0.0, 291.2}
#define HOTEND_OFFSET_Y {0.0, -0.18}
#define HOTEND_OFFSET_Z {0.0, 0.08}
and then fine tune through a host program like Pronterface.
A computer is not always available and it would be great to be able to fine tune using a menu item on the GLCD.

example images from repetier:-
[image]
[image]

thinkyhead, 2018-04-19:
Custom user menus can only be assigned a static G-code string.
What you're looking for requires writing code for a new menu.

marcio-ao, 2018-04-19:
At the moment, I think hot end offsets are hard-coded into Marlin at compile time. For an upcoming project, we want to have run-time configurable hot-end offsets. So I second this request.

thinkyhead, 2018-04-20:
> I think hot end offsets are hard-coded

They're not hardcoded, and can be changed at runtime by the `M218` command mentioned in the OP.

InsanityAutomation, 2019-10-15:
@boelle @shitcreek this one is done too

closed 2019-10-15 by boelle