MarlinFirmware/Marlin #10455: About the junction deviation function and the jerk calculation in planner.cpp

fatCat262, 2018-04-19:
Hello, I am a student who is currently working on the synchronization of a 3d printer with a robotic arm. To achieve synchornization, I need to at least calculate the parameters (v_entry, v_exit, v_nominal and most importantly the time of the block) of each segments in the planner. However, I found that the logic behind the planner (part about controlling the jerk) is quite abstract to read. Also, I am quite worry about that the motion control of Marlin which involves jerk (more accuratly the instant change of speed) which is not a smooth motion. It may cause problems when it comes to synchronize with other external mechanism. So I have some problems about the planner,

1. Is is possible to use junction deviation(blend radius method) which I have seen portion of the code in the planner.ccp to replace the jerk for a smooth motion? (which I think is the best case in terms of synchornization)

2. If the junction deviation function is not yet available, is there a way to predict the parameters (entry and exit speed and time interval of the segement) of the line segment. It is appreciated if I can get any advice for reading the code to understand the logic behind the jerk control portion? (As it is quite abstract to read)

3. To achieve synchornization of robot, (which its kinematics planner is usally using blending radius), I think I have to do points insertion between segements (breaking 1mm segment into maybe 1000 segments and input to marlin) to achieve desire velocity profile. Is it possible to do so? I know there is limiation on the frequency of the board and also the planner buffer size.

I konw probably many of maintainers are busy. Therefore, it is highly appreciated if I can get any of your advice. Thank you.

ejtagle, 2018-04-19:
We have Bézier velocity curve motion planning in bugfix2.0.0. Forget about breaking movements in so much segments. The CPU Marlin is using is simply not fast enough to process them.
The planner is already estimating v_initial, v_final, v_nominal and the execution time of the block, but i think it will be hard to keep synchronization over long periods of time due to rounding errors, and that sometimes Marlin could introduce delays in movements when it runs out of processing power.
Maybe your best choice would be not to use Marlin at all, and control everything yourself.
Look at the https://github.com/KevinOConnor/klipper project...

thinkyhead, 2018-04-20:
I'd like to have junction deviation (like Grbl) in the planner as an option so we can see how it works compared to the current jerk implementation. I started the process, but there's a lot of work yet to be done, since it touches several planner methods. Hopefully we'll have a testable version of it for Marlin 2.1.

fatCat262, 2018-04-20:
Thank you for your help, I will take a look into Klipper. Meanwhile, I think I will still try to break the segment into limited smaller segements in Marlin to see the effect. And I really look forward to the latest version of Marlin.

Squid116, 2018-05-02:
I'm working on a junction deviation implementation now.  It won't quite have all the benefits of the full grbl planner, but what I have so far is looking promising!

https://github.com/MarlinFirmware/Marlin/issues/10341

boelle, 2019-10-24:
@fatCat262 would it be wrong to close this one?

boelle, 2019-11-24:
closing, discussion can continue on discord

Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-11-24 by boelle