MarlinFirmware/Marlin #10457: [1.1.x-bugfix] Double diagonal moves with UBL enabled

gloomyandy, 2018-04-19:
### Description
Performing a diagonal move from (0, 270) to (220, 0) with UBL enabled works as expected but instead of stopping at (220, 0) the move "bounces" and heads back towards (0, 270). If you do not power off the printer the carriage will crash into the frame. With UBL disabled the move is as expected.

### Steps to Reproduce
Perform the following sequence...
G29 D // disable UBL
G28
G1 X0 Y270
G1 X220 Y0 // note move is fine
G29 L1 // load mesh
G29 A // Activate UBL
G1 X0 Y270
G1 X220 Y0 // initial move is fine but then the move "reverses" and heads back towards (0, 270)

**Expected behavior:** [What you expect to happen]
The move should be the same with and without UBL enabled

**Actual behavior:** [What actually happens]
See above the move is ok to begin with but instead of stopping it reverses direction 

#### Additional Information
My configuration files here:
https://github.com/gloomyandy/Marlin/tree/bugfix-1.1.x

Looking at the history for the file ubl_motion.cpp I noted that a recent change had commented out the following line 245:
`       //if (xi_cnt < 0 || yi_cnt < 0) break; // Too far! Exit the loop and go to FINAL_MOVE`
If I re-enable this test then things seem to work correctly (though I must admit the loop exit conditions seem a little strange, but I'm not familiar enough with this code to be able to suggest any other change).

Roxy-3D, 2018-04-19:
Thank You for finding this issue!!!    

I've reverted the logic to what it needs to be.

thinkyhead, 2018-04-20:
Aha! I commented that line for debugging. Thanks for putting it back!

gloomyandy, 2018-04-20:
Have tested the latest version all working fine for me now. Will close

closed 2018-04-20 by gloomyandy