MarlinFirmware/Marlin #10458: [bugfix-2.0.x] EEPROM datasize error

thisiskeithb, 2018-04-19:
### Description

Flashing the latest bugfix-2.0 as of yesterday resulted in an "EEPROM datasize Error" and I can no longer connect to my printer using OctoPrint.

I successfully flashed my board several times before this error using the same build of Marlin. I believe this error cropped up after I enabled #define LCD_SET_PROGRESS_MANUALLY
in Configuration_adv.h.

I have an MKS Gen-L 1.0 with 4 TMC2100 drivers. 

See [attached .zip](https://github.com/MarlinFirmware/Marlin/files/1929315/marlin_bugfix20_datasize_error.zip) for my config files & log output from OctoPrint.

thinkyhead, 2018-04-20:
> Flashing the latest bugfix-2.0 as of yesterday resulted in an "EEPROM datasize Error" and I can no longer connect to my printer using OctoPrint.

If it has an SD card slot, then make a G-code file containing:
```gcode
M502 ; default settings
M500 ; save settings
```
Save this file on SD and run it from the LCD controller. Maybe that will fix the EEPROM.

thisiskeithb, 2018-04-20:
I ran the .gcode file and power cycled the board. OctoPrint still reports the same error.

thinkyhead, 2018-04-20:
If you simply disable `EEPROM_SETTINGS` you're able to connect?

thisiskeithb, 2018-04-20:
I couldn't connect using OctoPrint, so I used a Windows machine with Arduino IDE. I was able to revert back to Marlin 1.1.8 & OctoPrint was able to connect fine. I recompiled the same build of bugfix-2.0.x that I was working from with `#define LCD_SET_PROGRESS_MANUALLY` commented out and OctoPrint connects fine.

thinkyhead, 2018-04-20:
Well, if you get a chance to flash `bugfix-1.1.x` with `EEPROM_SETTINGS` disabled, I'd be interested in the results.

thisiskeithb, 2018-04-20:
I ported my 1.1.8 settings (including `LCD_SET PROGRESS_MANUALLY`) to the latest `bugfix-1.1.x` branch with `EEPROM_SETTINGS` disabled and OctoPrint didn't complain about datasize. Attached are my config files. Enabling `EEPROM_SETTINGS` works as well.

[bugfix-1.1.x_config_files.zip](https://github.com/MarlinFirmware/Marlin/files/1930511/bugfix-1.1.x_config_files.zip)

thinkyhead, 2018-04-20:
Interesting. And yet, `bugfix-2.0.x` is still acting up?

thisiskeithb, 2018-04-20:
I spoke too soon. OctoPrint disconnected after my post flash GCODEs of M502 & M500 ran:

> Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin bugfix-1.1.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:TEVO Tornado EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:EEPROM:0
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: ok
Send: M21
Recv: echo:SD init fail
Recv: ok
Send: M117 IP 192.168.1.161
Recv: echo:SD init fail
Recv: ok
Send: M502
Recv: echo:Hardcoded Default Settings Loaded
Recv: ok
Send: M500
Recv: Error:EEPROM disabled
Changing monitoring state from "Operational" to "Error: EEPROM disabled"
Changing monitoring state from "Error: EEPROM disabled" to "Offline (Error: EEPROM disabled)"
Connection closed, closing down monitor

michaelwu21, 2018-04-20:
I'm having the same issue with the latest Marlin 2.0 on Mks sbase v1.3. Stock dvr drivers, config is basically the stock config without much changed.

https://pastebin.com/sYN4tBz9

Also, external sd card is still not working on sbase. Looking at spi_pins.h, the SCK and MISO pins are set to P0_15 and P0_17, which do not exist on EXP1

p3p, 2018-04-21:
>I'm having the same issue with the latest Marlin 2.0 on Mks sbase v1.3

@michaelwu21 if I remember correctly your issue is that the onboard sd-card is not working so you get eeprom data errors as Marlin cannot read the card, this is not related to any of the spi settings in Marlin. Most people seem to be able to get their onboard sd-cards working with the sbase if they format them, or change to another. This is something I want to investigate but I haven't found a card that doesn't work for me yet.

> Looking at spi_pins.h, the SCK and MISO pins are set to P0_15 and P0_17, which do not exist on EXP1

The pins specified in `spi_pins.h` file for the LPC1768x platform are defaults (and correct for the primary hardware spi port used on the Re-ARM), they are overridden by the boards pins file, `pins_MKS_SBASE.h` in your case. This shows that you need to use J8 and software spi, also it states you need to make a custom cable for controllers and external sd-cards.

The one error I can see in `pins_MKS_SBASE.h`, `#define SDSS  P0_06` should be `#define SDSS  P0_28` I've already told you about in #10174, also it really should work with `SDSUPPORT` disabled in the config, but you confirmed that did not fix the issue. I'm afraid its hard to debug something without the specific hardware and when for most people with it seems to work.

thisiskeithb, 2018-04-21:
To answer your question, @thinkyhead: Yes. With @benlye's comment, you can tell OcoPrint to ignore errors, but that's probably not a good fix.

michaelwu21, 2018-04-21:
Yeah I'm using ignore errors on octoprint which is how I got the terminal log(with the datasize error)

Still, I'd like to get eeprom to work- That was the main reason I updated to latest marlin, the older version I used simply would not do anything with the eeprom commands.

thinkyhead, 2018-04-21:
I don't see any obvious differences in comparing the two bugfix branches. So, here's something you can do to help track down the error. In `configuration_store.cpp` enable this option at the top of the file:
```cpp
#define DEBUG_EEPROM_READWRITE
```
Then when you boot the firmware and Marlin reads the EEPROM, it will report the first datafield where it finds a discrepancy between the expected offset and the actual offset. This will help narrow down the exact point where there's a mismatch.

thisiskeithb, 2018-04-21:
Here's the output from OctoPrint after flashing bugfix-2.0.x with `DEBUG_EEPROM_READWRITE` enabled, as well as `EEPROM_SETTINGS` and `LCD_SET_PROGRESS_MANUALLY`

https://pastebin.com/GiywKXZB

thinkyhead, 2018-04-22:
The logs show no errors. So I guess… congratulations…?

michaelwu21, 2018-04-26:
Huh. I tried enabling #define DEBUG_EEPROM_READWRITE, and surprisingly the datasize error disappeared.

However, there is no eeprom.dat saved on the sd card.

I did notice the following from octoprint output:
Send: N2 M21*18
Recv: echo:SD init fail
Recv: ok
Send: N3 M117 Octopi Connected*86
Recv: echo:SD init fail
Recv: Error:Line Number is not Last Line Number+1, Last Line: 0
Recv: Resend: 1

CaptCode, 2019-01-07:
not the best solution for this problem but at least the connection between the printer and octoprint is (and stays) established.

Octoprint:
Go to Settings -> Serial connections and select the Tab "Behaviour"
Under "Error handling" tick the box "Ignore".

Et voila!

Hope this helps...