MarlinFirmware/Marlin #10463: [Question] idle(), loop(), or lcd_update() seems to not get called on Archim 2.0?

marcio-ao, 2018-04-20:
I feel like I am missing something obvious, but on the Archim 2.0, a 32-bit board, I am finding that `idle()` isn't getting called.

Upon inspection, I see [this](https://github.com/MarlinFirmware/Marlin/blob/529c353c9116e74f8dd02831afc7570a75e26a16/Marlin/src/Marlin.cpp#L900):

```
void loop() {
  #if ENABLED(SDSUPPORT)
    card.checkautostart(false);
  #endif

  for (;;) {
    ...
  }
}
```

Isn't `for(;;)` essentially the same as `while(0) {}` which is dead-code? I tried adding `SERIAL_ECHOLNPGM("idle");` to the `loop() {}` and I get nothing, so it appears the `loop()` routine is no longer working? I still am able to interact with Marlin on the serial console, so I am quite confused as to how Marlin functions right now.

I can only suspect that in Marlin 2.0 there was some sort of architectural change I am not yet aware of. For context, my touch LCD code relies on `lcd_update()`, and it looks like that never gets called. Yet, I was able to get a RepRap Smart Controller to work, so again, I am perplexed as to the how.

Anyone willing to shed some light on what's going on with Marlin 32-bit?

marcio-ao, 2018-04-20:
Okay, so it looks like `for(;;)` is [idiomatic for loop forever](https://stackoverflow.com/questions/16113125/two-semicolons-inside-a-for-loop-parentheses), which I did not know. But that still does not explain why when I put a `SERIAL_ECHOLNPGM("idle");` inside the `loop()` I get nothing.

marcio-ao, 2018-04-20:
Okay, the plot thickens. If I get rid of code inside `loop()`, the watchdog timer fires and reboots the board periodically. So, it appears as if `loop()` is being called, but `SERIAL_ECHOLNPGM("idle");` isn't outputting anything.

Also, if I replace `loop()` with this:

```
void loop() {
  #if ENABLED(SDSUPPORT)
    card.checkautostart(false);
  #endif

  for (;;) {
    SERIAL_ECHO_START();
    SERIAL_ECHOPAIR(MSG_FREE_MEMORY, freeMemory());
    watchdog_reset();
    continue;
    ...
  }
}
```

I am still able to interact via the serial port and send commands like "M114", but yet I get no output from the  `SERIAL_ECHOPAIR()`. So it appears as if 1) the serial port and command processing do not rely on `loop()`, but more significantly, 2) that the serial routines do not operate in the `loop()` function. I'm a bit confused by all this.

p3p, 2018-04-20:
This isn't generalised to 32 bit, it is more likely an issue with Serial on the Due platform, or maybe even just the config on the Archim2.

marcio-ao, 2018-04-20:
@p3p: Okay. I've updated the title.

This is frustrating, it's hard to debug anything when your logging routines don't work :(

marcio-ao, 2018-04-20:
@p3p: Ahhh. Thanks for nudging me in the right direction. I found it:

```
/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT -1
```

I had SERIAL_PORT set to 0, but it seems to work when SERIAL_PORT is -1

closed 2018-04-20 by marcio-ao

p3p, 2018-04-20:
You stated earlier that you received serial output over the terminal when issuing g-code commands? Did you have both serial ports enabled? if so it should still have worked.

marcio-ao, 2018-04-20:
@p3p: SERIAL_PORT_2 was set to -1 by default, but it was not working with SERIAL_ECHOLNPGM. It only started to work when I set SERIAL_PORT to -1