MarlinFirmware/Marlin #10466: X max endstop stays open.

isaacfank, 2018-04-20:
### Description

I tried to do my homing and X max will not turn triggered. I home to max on my X axis. It works fine on 2.0 from October just tested, but on the newest zip download it does not trigger.

### Steps to Reproduce

1. Home.
2. Listen to it smash.
3. Sigh.

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-04-21:
It might be a problem with your configuration. Please ZIP and attach to a reply.

isaacfank, 2018-04-21:
Did this work?
Isaac Fankhauser
----------------------
IT/Network Admin
CNC Programmer
3D Design and Printing

---- On Fri, 20 Apr 2018 21:30:39 -0400 notifications@github.com wrote ----

It might be a problem with your configuration. Please ZIP and attach to a reply.

isaacfank, 2018-04-21:
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/1934066/configs.zip)
there we go.

thinkyhead, 2018-04-21:
What does `M119` report when the endstop is open, and what does it report when it is triggered?

isaacfank, 2018-04-21:
Oh sorry I forgot to mention that's how I troubleshooted it after it smashed into x. It stays open. So I moved x to triggered and loaded October and it says triggered. I then load todays and it goes to open. Even when away from the endstop it stays open. 
Isaac Fankhauser
----------------------
IT/Network Admin
CNC Programmer
3D Design and Printing

---- On Sat, 21 Apr 2018 05:35:59 -0400 notifications@github.com wrote ----

What does M119 report when the endstop is open, and what does it report when it is triggered?

isaacfank, 2018-04-21:
Y min and z min with probe works.

isaacfank, 2018-04-21:
[image]
I found it in the pins_RAMPS the pins are switched in the newest build? idk whats up with that.

p3p, 2018-04-21:
@isaacfank the default values for `X_MIN_PIN`(3), `X_MAX_PIN`(2) have not been changed .. well ever as far as I can see from the git history.

isaacfank, 2018-04-21:
Ooooo crap. I changed it. Downloaded 10/8 changed 10/9... Sorry. Closing.