MarlinFirmware/Marlin #10471: Ramps 1.4 with all TMC2130, dual Z steppers, only one is moving.

GizmoTheGreen, 2018-04-21:
### Description
I've got a Ramps board populated with TMC2130 drivers, I enabled Z_DUAL_STEPPER_DRIVERS in config as well as X, Y, Z, E0 and E1 as _IS_TMC2130
unless I understood it wrong this should automagically sync movement on Z and E1(as a Z2)
but that doesn't seem to be the case. 

M122 reports all drivers to have stealthChop enabled (they don't with 12v rail off, so I take this as a confirmation SPI comms are hooked up and working 100%)

### Steps to Reproduce

1. Enable X, Y, Z, E0 and E1 as _is_TMC2130
2. Enable Z_DUAL_STEPPER_DRIVERS
3. Compile and upload
**Expected behavior:**
When moving Z I expect both to move in unison

**Actual behavior:** 
Only the stepper hooked up to Z moves.

#### Additional Information
[Marlin config.zip](https://github.com/MarlinFirmware/Marlin/files/1934044/Marlin.config.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

GizmoTheGreen, 2018-04-21:
Attached M122 for what it's worth

Send: M122
Recv: 		X	Y	Z	E0	E1
Recv: Enabled		false	false	false	false	false
Recv: Set current	1000	1000	1000	800	1000
Recv: RMS current	994	994	994	795	994
Recv: MAX current	1402	1402	1402	1121	1402
Recv: Run current	17/31	17/31	17/31	25/31	17/31
Recv: Hold current	8/31	8/31	8/31	12/31	8/31
Recv: CS actual		8/31	8/31	8/31	12/31	8/31
Recv: PWM scale	1	1	1	1	1
Recv: vsense		0=.325	0=.325	0=.325	1=.18	0=.325
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold		0	0	0	0	0
Recv: [mm/s]		-	-	-	-	-
Recv: OT prewarn	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false
Recv: off time		5	5	5	5	5
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2	2
Recv: -start		3	3	3	3	3
Recv: Stallguard thrs	0	0	0	0	0
Recv: DRVSTATUS	X	Y	Z	E0	E1
Recv: stallguard
Recv: sg_result		0	0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:08:00:00
Recv: 	Y = 0x80:08:00:00
Recv: 	Z = 0x80:08:00:00
Recv: 	E0 = 0x80:0C:00:00
Recv: 	E1 = 0x80:08:00:00
Recv: 
Recv: 
Recv: ok

Ludy87, 2018-04-21:
Your settings are wrong and why is disable HYBRID_THRESHOLD?

## Configuration_adv.h
```
#define HAVE_TMC2130
#if ENABLED(HAVE_TMC2130)  // Choose your axes here. This is mandatory!
  #define X_IS_TMC2130
  //#define X2_IS_TMC2130
  #define Y_IS_TMC2130
  //#define Y2_IS_TMC2130
  #define Z_IS_TMC2130
  #define Z2_IS_TMC2130
  #define E0_IS_TMC2130
  //#define E1_IS_TMC2130
  //#define E2_IS_TMC2130
  //#define E3_IS_TMC2130
  //#define E4_IS_TMC2130
#endif
```
### pins_RAMPS.h

add this code
```
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN        -1 // Change PIN
#endif
```

thinkyhead, 2018-04-21:
Once you've figured out the right settings, be sure to test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch.

GizmoTheGreen, 2018-04-21:
Okay, thanks to the changes suggested by Ludy87 and thinkyhead I have the z2 axis working.
now I struggle with X axis not moving, it holds the motor in places but I can't move it... I think maybe the socket is fried, moving the TMC drivers around it's the only one not working and when it holds the motor it gets really hot.

anyway in regards to Z2 maybe have better comments on how it works in the config section and maybe leave the 
```
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN        -1 // Change PIN
#endif
```
inside pins_ramps.h

Ludy87, 2018-04-21:
Change the pinout according to your board.

-1 is a placeholder

GizmoTheGreen, 2018-04-21:
of course I did, I am not that dump ;)

it's working now, but I've managed to fry one of my tmc2130 boards, guess I'll use ordinary DRV8825 for extruder :/

many thanks to all for this awesome configurable/customizable fw

thinkyhead, 2018-04-22:
Frying boards is par for the course in RapRap-land!

closed 2018-04-22 by thinkyhead

BoKKeR, 2019-01-24:
Thanks for the help!

JayEnaR, 2019-06-30:
> you didn't define Z_DUAL_STEPPER_DRIVERS

Thanks for the help!